olofk/fusesoc #69: error when try to use fusesoc build atlys

KarlTao, 2015-01-26:
Hey there,
I am trying to build core for atlys board, and I got this error:

WARN:  File ../orpsoc-cores/systems/atlys/bench/orpsoc_tb.v does not exist
WARN:  File ../orpsoc-cores/systems/atlys/bench/uart_decoder.v does not exist

ERROR: Failed to build FPGA: Command 'xtclsh' not found. Make sure it is in $PATH

While I have sourced setting32.sh and can run xtclsh directly from terminal.
Do you have a clue what is this issue about?
Thanks!
Karl

olofk, 2015-01-27:
This is strange. The warnings are harmless, but I can't understand why it can't find the tools if you have sourced the correct settings file. I just tried it myself and don't have this problem.

Could you try to run the following lines as a python script to see if you get an error

import subprocess
subprocess.check_call(['xtclsh', '-h'])

KarlTao, 2015-01-27:
Hi,
I ran the command you said, and here is the result:

Release 14.7 - xtclsh P.20131013 (lin)
Copyright (c) 1995-2013 Xilinx, Inc.  All rights reserved.
xtclsh                                - Invoke tclsh in a interactive mode.
xtclsh -h                             - Print out the usage and version info and
exit.
xtclsh <Tcl_script> [arg1] ... [argn] - Evaluate the Tcl script with the
optional arguments and exit.

olofk, 2015-01-27:
ok, that's odd. This is basically what fusesoc does internally. I don't have a clue for what's wrong. Just in case, you could remove the PATH lines from .bashrc. They shouldn't be needed if you source settings64.sh correctly

What OS and python version are you running?

KarlTao, 2015-01-27:
Hi,
Thank you for replying, I solved it by running fusesoc init again, it downloaded some additional files. But it is odd since I have tried to build it on several different computers, and none of them worked after running fusesoc init once...
Anyway, Thank you very much for your answer.

olofk, 2015-01-27:
Happy to hear that it worked out in the end even though I still don't know what was wrong.

I'm closing this now, but will check for potential problem with the init command, and also see if there are any other problems that could lead to the error message that you got. Thanks for reporting

closed 2015-01-27 by olofk