olofk/fusesoc #72: ERRO 'xtclsh'

suelin, 2015-02-24:
Hello,

   So, I've been trying to build atlys with the command, '../fusesoc/bin/fusesoc build atlys', but he gives this erro, 'ERROR: Failed to build FPGA: Command 'xtclsh' not found. Make sure it is in $PATH'.
   I don't know what can I do, I searched a lot but nothing help me.
   Do you know what is it?

olofk, 2015-02-24:
Have you source the Xilinx settings script? If not, they are located in your ISE root directory, so do something like "source /opt/Xilinx/14.6/ISE_DS/settings64.sh" (adjust according to your installation directory and use settings32.sh for a 32-bit machine. 

If you already have sourced the settings file, I'm not sure what it could be without some extra debugging. There was a similar issue a while ago reported here https://github.com/olofk/fusesoc/issues/69

olofk, 2015-11-11:
Closing this due to lack of activity.

closed 2015-11-11 by olofk