olofk/fusesoc #73: Error build atlys

suelin, 2015-02-26:
Hello!
    This problem occurs when I try to build atlys.
    When I check the .cache/fusesoc/* after the error, some of the directories are empity.

WARN:  File /root/.cache/fusesoc/adv_debug_sys/rtl/verilog/adbg_wb_biu.v does not exist
WARN:  File /root/.cache/fusesoc/adv_debug_sys/rtl/verilog/adbg_or1k_status_reg.v does not exist
WARN:  File /root/.cache/fusesoc/adv_debug_sys/rtl/verilog/adbg_jsp_biu.v does not exist
...
INFO:  Checking out http://github.com/openrisc/mor1kx/archive/master.tar.gz revision master to /root/.cache/fusesoc/mor1kx
http://github.com/openrisc/mor1kx/archive/master.tar.gz
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 302, in <module>
    run(parsed_args)
  File "/usr/local/bin/fusesoc", line 242, in run
    args.func(args)
  File "/usr/local/bin/fusesoc", line 51, in build
    backend.configure()
  File "/usr/local/lib/python2.7/dist-packages/fusesoc/build/ise.py", line 34, in configure
    super(Ise, self).configure()
  File "/usr/local/lib/python2.7/dist-packages/fusesoc/build/backend.py", line 57, in configure
    core.setup()
  File "/usr/local/lib/python2.7/dist-packages/fusesoc/core.py", line 98, in setup
    if self.provider.fetch():
  File "/usr/local/lib/python2.7/dist-packages/fusesoc/provider/github.py", line 26, in fetch
    self._checkout(self.files_root)
  File "/usr/local/lib/python2.7/dist-packages/fusesoc/provider/github.py", line 47, in _checkout
    (filename, headers) = urllib.urlretrieve(url)
  File "/usr/lib/python2.7/urllib.py", line 93, in urlretrieve
    return _urlopener.retrieve(url, filename, reporthook, data)
  File "/usr/lib/python2.7/urllib.py", line 239, in retrieve
    fp = self.open(url, data)
  File "/usr/lib/python2.7/urllib.py", line 207, in open
    return getattr(self, name)(url)
  File "/usr/lib/python2.7/urllib.py", line 344, in open_http
    h.endheaders(data)
  File "/usr/lib/python2.7/httplib.py", line 954, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 814, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 776, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 757, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 553, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
IOError: [Errno socket error] [Errno -2] Name or service not known

olofk, 2015-02-27:
Hi,

I think the problem is that there are problems with the OpenCores SVN server, so FuseSoC can't download the cores. I'm working on migrating all the cores to fetch from github instead. I will let you know when that is done

olofk, 2015-11-11:
Was this resolved?

olofk, 2016-07-18:
Closing this due to inactivity. Feel free to reopen if it's still a problem

closed 2016-07-18 by olofk