olofk/fusesoc #77: Implement some way to at least warn about the system packages required by the HDL packages

mgielda, 2015-10-13:
The example in the README, `wb_sdram_ctrl`, actually requires development version of `libelf` (`libelf-dev` on Ubuntu) but if you don't have that, you will only see:

```ERROR: Failed to build simulation model
ERROR: Failed to compile VPI library elf-loader```

In an e-mail discussion with @olofk he mentions he is aware of that but hasn't yet figured out a good way to handle this kind of stuff - since this is not a FuseSoC requirement per se - so I proposed an issue to initiate the discussion.

The least we could do of course would be to enable some messages to be printed by the package at build (like: `This package requires the development version of libelf, which may be libelf-dev on your system. Make sure you have it installed.`). I use that in our buildsystems where it is not trivial to explicitly check for something: of course this is not idea, but at least the developer would not need to bother people with questions if he can spot what's wrong himself.

Just writing it down in want of other propositions so far, I am sure there are better ways to handle it. Just thinking that actually having an example that can fail without a unambiguous warning/error message could potentially stop people who do 'let's download this and see what it is - not working - not useful' from giving FuseSoC a try. I happen to do exactly that if I'm not determined, the bad person I am ;)

olofk, 2015-10-13:
There might be a simple way to solve this, even though it's not the prettiest way. FuseSoC has support for running scripts at different stages. pre_run_scripts are executed before the simulation is run, so we could put in a script that checks for gelf.h.

Unfortunately, right now I think that only scripts in the top-level core is run, but that should be easy to fix, so that scripts from all the included cores are run.

If we get a lot of external dependencies, we should probably spend some effort to make a better mechanism for this, but I think this might actually be one of the only external requirements so we don't have to solve it in a completely generic way

mgielda, 2015-10-14:
This sounds like a good enough idea for now! If there is already a good mechanism that will work, no sense inventing a new one before we hit further problems, which probably will not happen any time soon.

I can see `pre_run_scripts` was not implemented in any of the cores in the main repo, am I right? If you could implement the solution you mention I could document how the `pre_run_scripts` section can be used for things like checking dependencies.

olofk, 2015-10-15:
I have now made some updates to FuseSoC so that scripts are run for all cores in the dependency lists as well. Now we just need to add the following to elf-loader.core:
> [scripts]
> pre_run_scripts = check_for_libelf.sh

and add an executable script called check_for_libelf.sh with something like this in it:
> echo \#include \<gelf.h\> > test.c
> echo void main\(void\) \{\} >> test.c
> 
> gcc test.c || echo "Error: Missing libelf headers"

Not sure what is best practice to search for this, so I'm leaving the last part for someone else to implement.

Note that two more things should be improved as well
1. This only applies to simulations. I realized that the scripts executed in the FPGA build flow are fetched from the system file. Need some refactoring here
2. Need to break or warn on script exit code. Not sure about best practice here either. Perhaps 0 = success, >0 = warning,  <0 = error or something like that. Love to get some input here as well.

olofk, 2015-10-18:
FuseSoC now aborts on scripts that returns an exit code != 0. Now we just need add the scripts for the missing dependencies

olofk, 2015-10-18:
Oh well. I added and pushed the scripts as well. Please try it out and close the issue if it works

mgielda, 2015-10-18:
Awesome! Works. Just fixed a little bug in https://github.com/olofk/fusesoc/pull/81

Closing this one.

closed 2015-10-18 by mgielda