olofk/fusesoc #82: Fusesoc for Actel ProASIC3

salmansheikh, 2015-12-11:
What changes needed to use this for an Actel ProASIC3? Which modules need customization?

olofk, 2015-12-12:
This would require a new backend module for Libero similar to the ISE and Quartus files here https://github.com/olofk/fusesoc/tree/master/fusesoc/build

salmansheikh, 2015-12-12:
But what cores need to be custom for the ProASIC? I think there are subtle nuances between xilinx and altera beyond just the tools called and pins, right? Or are all the cores the same exactly?

olofk, 2015-12-12:
That's hard to tell, but hopefully, we just need a new top-level core. We used to build against a ProASIC3 in orpsocv2, and I don't think we did anything special for the cores we used there

olofk, 2017-10-10:
Closing due to inactivity

closed 2017-10-10 by olofk