olofk/fusesoc #83: unknown module <pfpu32_top>

suelin, 2015-12-22:
Hi, 
I'm using the fusesoc and I need to use the mor1kx with FPU, I found this posted here: https://github.com/openrisc/mor1kx.
I can't use this mor1kx on the fusesoc because when I synthesize the project on ISE, I have this error: 
HDLCompiler:1654 - "/home/orpsoc/Documentos/Fusesoc/fusesoc/build/atlys/src/mor1kx-3.1/rtl/verilog/mor1kx_execute_alu.v" Line 512: Instantiating <u_pfpu32> from unknown module <pfpu32_top>
Can you help me?

olofk, 2015-12-22:
Hi. This sounds strange. The FPU wasn't added to mor1kx until version 4.

It looks like you are trying to build the atlys system, so my recommendation is to change the dependency on mor1kx-3.1 to mor1kx-4.1 in atlys.core

suelin, 2015-12-23:
That's it, now is working
Thank you so much!!

closed 2015-12-23 by suelin