olofk/fusesoc #84: Use Python distutils for installation

chiggs, 2015-12-27:
The module installation is a little overkill and unfriendly to Python users.  Since there is nothing exotic that needs building, I suggest migrating to the standard python `setup.py` installation script.

This would then enable the project to be placed on PyPi and users could then install using the typical Python mechanism: `pip install fusesoc`.

olofk, 2015-12-27:
The original plan was to use distutils, distribute, setuptools, pip, easy_install, waf, scons... whatever it's called, but Python packaging looked like a mess to me, so I decided that it was way easier to use autotools instead. One thing that I couldn't figure out was how to install the package to some user-writable location. It seemed like it forced installation to /usr/lib/.... I'm not sure I'm convinced by the greatness of pip/pypi either when at least all Linux distributions already have package managers.

With that said, I'll happly accept patches however to make it work in a more python-friendly way. I would however prefer to have the automake system still around side-by-side with distutils, at least until the next released version

olofk, 2015-12-27:
I tried the patch now and installing works fine. There are some things that I can't figure out how to do however.
1. I can't see any way to uninstall
2. I'm having problems installing to a custom directory. (Not sure I need this though. Can as well run from the source directory)
3. Can't figure out how to handle dependencies. Right now there's a hard dep on git and some other basic tools. Also, svn is an optional dependency. (In worst case these could be handled at runtime with exceptions, but having proper install-time dependencies makes it easier for package maintainers)

chiggs, 2015-12-27:
Agreed, Python packaging is a total mess.  Sadly, it's probably never going to get sorted out.  Question is whether to try and fit in with the mess or not - I think it will be easier for users ultimately if `fusesoc` behaves like other Python packages.  Keeping automake in parallel with distutils sounds like a very sensible compromise.

> I can't see any way to uninstall

This is a general problem with Python packaging.  In theory `pip uninstall fusesoc` should work though I haven't tested it.

> I'm having problems installing to a custom directory. (Not sure I need this though. Can as well run from the source directory)

```
pip install --install-option="--prefix=/some/user/location" .
```

> Can't figure out how to handle dependencies.

Apparently the standard mechanism is to check in `setup.py` and raise an exception with a helpful error message.  This doesn't help package maintainers though and is a slightly cruddy way of doing things.

olofk, 2015-12-27:
Fair enough. I see the benefits of being on pypi and keeping up with what the rest of the Python world is doing. I simplified the autotools files a bit to make it easier to co-exist with distutils. If you rebase your patches, just add setup.{cfg,py} and change version to 1.4-dev (or whatever distutils uses to indicate that this is a development version that will end up as 1.4 in time) I'm happy to add this

commit 8a901e3f (2016-01-04): Fixes #84: Migrate to distutils and setup.py
closed 2016-01-04 by olofk