olofk/fusesoc #87: simulation problem

aliahari, 2016-03-18:
Hi,
when i try to do a simple simulation by calling 
fusesoc sim --sim modelsim wb_sdram_ctrl

I get the the following error:
  File "build/bdist.linux-x86_64/egg/fusesoc/simulator/modelsim.py", line 24, in configure
  File "build/bdist.linux-x86_64/egg/fusesoc/simulator/simulator.py", line 67, in configure
  File "build/bdist.linux-x86_64/egg/fusesoc/edatool.py", line 100, in parse_args
KeyError: 'elf_load'

I have tried also using other cores and also icarus simulator but I always get the same error.

Any idea?

olofk, 2016-03-18:
Oops. You're right. Thanks for reporting. I pushed a fix now. Please close the bug if it works or let me know if the problem remains

aliahari, 2016-03-21:
Thanks. The issue is resolved.

closed 2016-03-21 by aliahari