olofk/fusesoc #88: simulation problem with questasim (modelsim)

aliahari, 2016-03-21:
Hi, 

when I am using icarus as the simulator everythin is fine but when I try it with questasim (--sim=modelsim) which has the same interface and commands as modelsim I got the following error:

Running: .../check_libelf.sh
ld: cannot find -lelf
ERROR: Failed to build simulation model
ERROR: Linking of elf-loader failed

However, the libelf-dev package is already installed and the MODEL_TECH evn. variable is set.

olofk, 2016-03-21:
hmm... I can't reproduce it here, and I don't have access to questasim. I wonder if this is a 32/64-bit issue. The free modelsim version I'm using only support 32-bit. There are some things we could try:

1. Assuming that you're simulating wb_sdram_ctrl, what happens if you cd into "build/wb_sdram_ctrl/sim-modelsim" and run "ld -shared -E -melf_i386 -o e.vpi elf-loader.o -lelf" ? Try it both with and without the -melf_i386 parameter

2. Set the MTI_VCO_MODE env var to 32. (export MTI_VCO_MODE=32) before running

aliahari, 2016-03-21:
Exactly I think this is related to a 32/64 bit mismatch. Questasim is somehow the 64-bit version of modelsim. Here are the outcome of the above mentioned actions.
1. with -melf_i386 it again gives the error "ld: cannot find -lelf", however without this option it gives the message "ld: i386 architecture of input file `elf-loader.o' is incompatible with i386:x86-64 output" then it says that in elf-loader.c functions like close and puts are not defined. 
2. when I set this variable it could not find the vlib since it looks for it in a wrong address
$PATH_TO_QUESTA_BIN/../linux/vlib

I have tried also to run fusesoc with --64 option but nothing is changed.

olofk, 2016-03-21:
Thanks. That's good to know. I don't have a ready fix for this unfortunately, but at least now I know what to do. You could try to remove the lines `args += ['-m32']` and `args += ['-m32']` from `modelsim.py` and see if that helps.

olofk, 2019-01-15:
I hope this is now finally fixed with https://github.com/olofk/edalize/commit/a8e97f8c6090e9b34ce00211b3fee06a8c0591b7

Please reopen otherwise

closed 2019-01-15 by olofk