olofk/fusesoc #90: Handle different SoCs with different boards

wallento, 2016-03-26:
Hi,

I was wondering if there is a plan to add the concept of a board. With the OpenRISC background, currently a system is more or less a board. I plan to have a different SoC now and have as much reuse as possible.

Here is what I am doing now:
* A core `<SoC>` for the SoC toplevel that has a "generic" interface
* A core `<Board>` that contains the board-specific blocks, like the DDR controller etc.
* Create a system: `<SoC>-<Board>` that has a glue logic toplevel

I think it will be complex to find a really generic way.

Best,
Stefan

olofk, 2016-03-26:
I have thought a lot about this too, and I don't think it's feasible to do it totally generic. The goal has instead been to make the system as light-weight as possible. Unfortunately we aren't really there yet. Some systems have a lot of `ifdef code to make it slightly configurable, but I'm working on removing this as it makes the systems pretty messy instead. It gets increasingly more complicated as there currently isn't any way to bring in just a subset of the constraints, which means that all constraints has to be valid for all combinations.

In the future, I hope to push some of the boring tasks, such as connecting the top-level cores into the IP-XACT domain. That will make it a bit faster to create new variants of a SoC for a board. For the OpenRISC systems also, I have thought of making a basic core containing CPU+UART+GPIO+and possibly something more such as a SPI controller, and just having board-specific cores depend on this. From what I understand, this is quite similar to your approach.

For now, I quite like the method used here https://github.com/andrzej-r/orpsoc-cores/tree/nexys4ddr/systems/nexys4ddr to have a collection of board-specific cores, that can be used by several systems using this board.

wallento, 2016-03-27:
Yes, I have picked this style up now and extended it a bit to encapsulate everything board-specific in one core that is then sourced by the system.

wallento, 2016-03-28:
Okay, I drafted it a bit here: https://github.com/wallento/fusesoc_cores

The "board abstraction module" `arty` contains all core-specific stuff, there will be a `mor1kx-soc` as the SoC toplevel (wip). In `mor1kx-arty` it is more or less just some wires and tie-offs. I will play this through for a few board.

Note: Testing this requires #93 and #92

wallento, 2016-04-16:
This is something that can be done with fusesoc, I would not address it for now by some special types or so.

closed 2016-04-16 by wallento