olofk/fusesoc #91: Unify system and cores root and change to providers style

wallento, 2016-03-26:
Hi,

I am wondering why `system_root` and `cores_root` are distinct values. Shouldn't they be unified as one kind of "FuseSoC Catalogs".

Beside this I would suggest to extend those from file system directories to also some kind of provider.

Proposal: Add a `fusesoc_catalogs` variable to the config (and/or `FUSESOC_CATALOGS` to the environment), that is a colon-separated list of:

* Filesystem paths, format: `/path/to/catalog`
* http path, format `https://github.com/olofk/coolip/master/tree/`
* any other provider. I am thinking of a librecores.org API that can be used as a provider, so that `librecores` can be a catalog.

Once starting in this direction a more sophisticated dependency format should be possible. It should ideally allow for stuff like `librecores:openrisc/mor1kx@v4.1` or some other cool stuff. But thats probably another issue then.

Best,
Stefan

olofk, 2016-03-26:
Thanks for the proposal. It's great to have someone else thinking about this as well. Makes it more like a community effort than a one-man show :)

> I am wondering why system_root and cores_root are distinct values. Shouldn't they be unified as one kind of "FuseSoC Catalogs".

system_root is deprecated since a while back. I thought I had removed the last trace of that already. Oh.. now I see that it's even there in the shipped fusesoc.conf. I'll fix that

> Proposal: Add a fusesoc_catalogs variable to the config (and/or FUSESOC_CATALOGS to the environment), that is a colon-separated list of:
>
> * Filesystem paths, format: /path/to/catalog
> * http path, [...]

Adding the possibility to load core libraries (which is what I call the fusesoc_catalogs in your proposal) from a http address is an interesting idea (even though you might want to rethink having : as a separator for web addresses :))

I'd like to extend this a bit however and add a separate section for each library. That would also make it possible to add more features in the future, such as enabling or preventing automatic upgrades, adding dependencies between libraries and so on. It could perhaps look something like this

    [library orpsoc_cores]
    description=Basic core library for FuseSoC
    type=git
    auto-update=true
    
    [library evil_cores]
    description=Collection of cores used for evil things
    type=webdav
    depend=orpsoc_cores

wallento, 2016-03-28:
Yes, good starting point for a discussion. I will draft a simple web API with static data the next weeks and we should take it from there.

olofk, 2018-01-17:
The library part that I described above is implemented with https://github.com/olofk/fusesoc/pull/186

imphil, 2020-12-18:
There's nothing actionable left in this issue, is there?

closed 2021-02-12 by imphil