olofk/fusesoc #97: Add environment variable FUSESOC_ROOT

wallento, 2016-04-15:
Hi,

it would be great if fusesoc did not only search for cores in the configured cores_root, but if the user can add further roots in an environment variable.

Best,
Stefan

olofk, 2016-04-15:
Not particularly fond of environment variables. We can already add --cores-root=/path/to/library to add additional libraries. The cores in the libraries specified this way will take precedence over those found in fusesoc.con

wallento, 2016-04-15:
Yeah, but people may want to overlay cores with their own implementation without changing Makefiles. Thats the situation we have in OpTiMSoC now. We can circumvent this by adding an OPTIMSOC_CORES environment variables and supply this to --cores_root, but I think its like a standard feature to support the environment, commandline and configuration precendence.

olofk, 2016-04-15:
If it makes things easier, we could add it, but as you say, if FuseSoC is run from a script, you could set --cores-root=$VAR there.

If we were to add it, I would propose that the order would be command line > env var > fusesoc.conf, and that we supply it as a colon-separated list (as $PATH). Is that fine with you?

wallento, 2016-04-15:
Yes, you are right, I got the order wrong. I will have a look this feature, because I want to understand this part better (wrt librecores later). Cheers!

olofk, 2016-09-23:
Fixed in 74dc8e274be683688329b369abe31ab1a63ba0a6

closed 2016-09-23 by olofk