olofk/fusesoc #98: Plot the dependency graph with graphviz

wallento, 2016-04-16:
The idea is to plot a dependency graph for the given core/system

olofk, 2016-04-16:
I LOVE it! Where's the code? :)

wallento, 2016-04-16:
Here: #99 :)

wallento, 2016-04-16:
[image]

m-kru, 2021-01-21:
@olofk  What is the status of generating dependency graph functionality? It would be useful for working on parallelization and caching analysis results as described in https://forums.ohwr.org/t/hdlmake-vs-fusesoc/848312/8.

imphil, 2021-01-21:
There are patches to get such a graph in the OpenTitan version of fusesoc, you can install it with `pip3 install git+https://github.com/lowRISC/fusesoc.git@ot#egg=fusesoc`. This branch depends on a couple other things which need to go upstream first in a "proper" way (we do some things there which break backwards-compatibility).

m-kru, 2021-01-21:
@imphil How is OpenTitan version FuseSoc different than FuseSoc? Any why are there more than one FuseSoc version?

imphil, 2021-01-21:
It's just a version with patches which are not yet upstream, you can find the git repository at https://github.com/lowRISC/fusesoc

closed 2025-02-21 by olofk