olofk/fusesoc #101: Install error - ghdl

Hooligan-0, 2016-04-18:
Install for new users fails. With the current version (installed using "pip") GHDL is not available but opa core into current repository try to load it. So, after install, all commands just show:
ValueError: Invalid value 'ghdl'. Allowed values are 'icarus', 'modelsim', 'verilator', 'isim', 'xsim'

olofk, 2016-04-19:
Yeah, this is not good. I think I will have to drop the ghdl support from opa until I have released a new stable version of FuseSoC. Thanks for reporting

olofk, 2016-04-19:
I pushed https://github.com/openrisc/orpsoc-cores/commit/f2716132a2e73a4cecd68ef00c4dd0f0e2d97dc9 to take care of it for now. I'll also patch FuseSoC to handle unknown simulators better in the future

olofk, 2016-04-19:
And here's the fix on the FuseSoC side https://github.com/olofk/fusesoc/commit/52965ccf5e47fd3d8607c83e01f084de1f5b8127 Please let me know if this works and I'll close the issue

Hooligan-0, 2016-04-19:
It works for me ! :)

olofk, 2016-04-19:
That's all I need to hear. :)
Thanks for reporting and testing!

closed 2016-04-19 by olofk