olofk/fusesoc #102: simulation of mor1kx leads to compilation error

flenkium, 2016-06-13:
Hello,

I want to simulate the mor1kx module and the new branch predictor files (mor1kx_branch_predictor_simple.v) leads to a compilation error:

```
$$ fusesoc sim mor1kx-generic
...
INFO: Preparing mor1kx
INFO: Downloading openrisc/mor1kx from github
...
INFO: Preparing mor1kx-generic

INFO: Running /home//.local/share/orpsoc-cores/cores/elf-loader/check_libelf.sh
Compiling /home//workspace/openrisc/exe/build/mor1kx-generic/src/elf-loader/elf-loader.c...
Compiling /home//workspace/openrisc/exe/build/mor1kx-generic/src/elf-loader/vpi_wrapper.c...
Making elf-loader.vpi from elf-loader.o vpi_wrapper.o...
Compiling /home//workspace/openrisc/exe/build/mor1kx-generic/src/jtag_vpi-r2/jtag_vpi.c...
Making jtag_vpi-r2.vpi from jtag_vpi.o...
../src/mor1kx/rtl/verilog/mor1kx_branch_predictor_simple.v:19: syntax error
I give up.
ERROR: Failed to build simulation model
ERROR: Failed to compile Icarus Simulation model
```

This error also occurs, if I load an ELF-file

Do I something wrong?

olofk, 2016-06-13:
I found the problem now. A recent commit to mor1kx added some code that Icarus Verilog can't parse. I've sent in a pull request to fix it. As soon as that is accepted, it should work again. You can track the progress here https://github.com/openrisc/mor1kx/pull/35

flenkium, 2016-06-13:
Thank you for your fast answer. :-)

closed 2016-06-13 by flenkium