olofk/fusesoc #104: windows installation problems

nturley, 2016-06-26:
I ran into a few bumps trying out fusesoc, this is what I tried on a windows 10 machine with python 2.7.8
```
>pip install fusesoc
...
Successfully installed fusesoc-1.4
>fusesoc init
'fusesoc' is not recognized as an internal or external command,
operable program or batch file.
```
looks like setup.py put a file named fusesoc in C:/Python27/Scripts. Let's try running that.
```
>python C:\Python27\Scripts\fusesoc init
Traceback (most recent call last):
  File "C:\Python27\Scripts\fusesoc.py", line 25, in <module>
    from fusesoc.build import BackendFactory
  File "C:\Python27\Scripts\fusesoc.py", line 25, in <module>
    from fusesoc.build import BackendFactory
ImportError: No module named build
```
It's confused. I decided to try inside of a virtual environment.
```
>python venv/Scripts/fusesoc init
...
ImportError: No module named yaml
```
That got further. That one looks like a missing dependency
```
>pip install pyyaml
>python C:\Python27\Scripts\fusesoc init
...
INFO:  FuseSoC is ready to use!
```

nturley, 2016-06-26:
I submitted a PR for this issue

olofk, 2016-06-26:
Thanks. I've seen the PR. FYI, I love to have this working on windows, but I'm not sure anyone has tested it that much, so I expect you to hit some problems after it's installed. I'm happy to review and discuss any windows-related patches, but as I don't have easy access to a windows machine, I can't really test things myself that easily

nturley, 2016-06-26:
Cross platform testing is a pain. Usually that's what CI is for. Have you thought about using travis or appveyor? I'll spend some time looking into it and you can see if you like it.

olofk, 2016-06-26:
Oh, that would also be very welcome. I've been kicking myself for not doing unit testing from the beginning. Could have caught a lot of unnecessary errors that way. But just build/install testing with travis (or similar) would be great.

nturley, 2016-06-29:
I broke this up with #106, #107, and #108. So at this point this issue is scoped to windows not being able to execute the "fusesoc" file in C:/Python27/Scripts.

olofk, 2016-07-09:
Hopefully this should be fixed now. Would be great if you or someone else could confirm this. Thanks!

nturley, 2016-07-09:
Awesome!

closed 2016-07-09 by nturley