olofk/fusesoc #107: Add system level tests to CI

nturley, 2016-06-29:
In addition to a basic smoke test (#106), the CI should also exercise building, simulation, and synthesis.

To avoid the orpsoc-cores repo breaking these tests, a dummy test repo should be created. Just enough to exercise all of the fusesoc features.

To get good coverage, calls to 3rd party proprietary tools should probably be mocked. Open source simulators and synthesizers (icarus, ghdl, icestorm) can be installed and run from the ci.

olofk, 2017-03-14:
I've started working on this now. Not system tests, but unit tests with mocked external commands

oleg-nenashev, 2017-03-14:
I also had several proofs of concept on LibreCores CI, but have never reached the releasable state. And IIRC was going to provide examples to @olofk and then dropped the ball on it :(

oleg-nenashev, 2017-03-20:
@olofk Feel free to assign it to me, work-in-progress

olofk, 2017-03-21:
Feel free to take it. Can't figure out how to assign with the github gui though