olofk/fusesoc #111: windows support -- requires patch command

nturley, 2016-06-30:
Windows systems are unlikely to have a commandline patch utility installed. So building any cores with patches will fail with

`Error: Failed to call external command 'patch'`

If the call to patch is replaced with native python that will bring fuse-soc closer to supporting windows

nturley, 2016-06-30:
Alternatively you could force windows users to use mingw or cygwin or something. It looks like a some of the packages invoke gcc or shell scripts anyway.

olofk, 2016-06-30:
Yes, I have tried to minimize the number of calls to external tools to make it more portable. Replacing the call to patch with a native python tool has been on my agenda for a few years now, but as there haven't been that many windows users yet, I've been pushing it. I think I found some implementation of patch on Google code some time ago which had some limitations, but maybe the situation is better now.

However, as you mention, some of the cores need to run scripts at some stages too. I could have required this to be python scripts, but I don't want to have a python dependency in the .core files (even though FuseSoC is written in Python, other tools dealing with .core files might not be). Another option is, as you say, to run this in cygwin or similar. Problem there is I think that it's hard to launch native windows EDA tools from that environment. Although, I could be wrong here. Don't know enough about windows unfortunately. I've also been thinking that it might work out just fine now when Windows has some kind of native Linux syscall wrapper thing in Windows 10. Would it help to use that?

nturley, 2016-07-01:
it still doesn't look super good IMO, but I found a couple that might work. I found one named python-patch which I really liked, the only limitation I ran into was that it was incapable of creating or deleting files (by design, because of the security implications). Now that you changed that one patch to do a file instead, it probably would work perfectly. I'll get to work on it.

olofk, 2016-11-22:
I changed to using git-apply instead of patch, as git is already a required dependency. Works fine on Linux, but I don't want to push this to master until someone tried it on windows as well. Any volunteers? :)

olofk, 2016-11-22:
Oh, the code is on [this branch](https://github.com/olofk/fusesoc/tree/git-apply)

olofk, 2016-12-23:
I have now tested this myself on a windows platform and 742068c844b7977febc338267b5990e88d5f406d removes the need for an external `patch` command

closed 2016-12-23 by olofk