olofk/fusesoc #118: "fusesoc build" returns zero result code when synthesis fails

oleg-nenashev, 2016-09-08:
Just a follow-up to https://github.com/openrisc/orpsoc-cores/issues/105

When Analysis and synthesis fails for "fusesoc build xxx", fusesoc prints errors to the console, but then exits with the 0 exit code. It may confuse CI, because it requires special handling on it's side.

It happens due to the missing <code>exit()</code> call in https://github.com/olofk/fusesoc/blob/8c1bec7997855a8a3f97e6d1a15464ac6bfb7c23/fusesoc/main.py#L85-L86 . Since the same behavior happens for other commands like <code>pgm</code>, maybe this behavior is "as designed".

```
Error (10170): Verilog HDL syntax error at orpsoc_top.v(915) near text: ")"; mismatched closing parenthesis . Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Altera Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number. File: /fusesoc/build/sockit/src/sockit/rtl/verilog/orpsoc_top.v Line: 915
Error (10170): Verilog HDL syntax error at orpsoc_top.v(915) near text: ")";  expecting "(". Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Altera Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number. File: /fusesoc/build/sockit/src/sockit/rtl/verilog/orpsoc_top.v Line: 915
Error (10112): Ignored design unit "orpsoc_top" at orpsoc_top.v(36) due to previous errors File: /fusesoc/build/sockit/src/sockit/rtl/verilog/orpsoc_top.v Line: 36
Info (12021): Found 0 design units, including 0 entities, in source file /fusesoc/build/sockit/src/sockit/rtl/verilog/orpsoc_top.v
Info (12021): Found 1 design units, including 1 entities, in source file /fusesoc/build/sockit/src/sockit/rtl/verilog/rom.v
    Info (12023): Found entity 1: rom File: /fusesoc/build/sockit/src/sockit/rtl/verilog/rom.v Line: 36
Info (12021): Found 1 design units, including 1 entities, in source file /fusesoc/build/sockit/src/sockit/rtl/verilog/wb_intercon.v
    Info (12023): Found entity 1: wb_intercon File: /fusesoc/build/sockit/src/sockit/rtl/verilog/wb_intercon.v Line: 1
Info (144001): Generated suppressed messages file /fusesoc/build/sockit/bld-quartus/sockit.map.smsg
Error: Quartus Prime Analysis & Synthesis was unsuccessful. 3 errors, 8 warnings
    Error: Peak virtual memory: 1147 megabytes
    Error: Processing ended: Thu Sep  8 12:29:15 2016
    Error: Elapsed time: 00:00:11
    Error: Total CPU time (on all processors): 00:00:18
Makefile:11: recipe for target 'map' failed
make: *** [map] Error 3
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in elf-loader[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_sdram_ctrl[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_altera_ddr_wrapper[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_intercon[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_intercon-1.0[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in fifo-1.0[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in vlog_tb_utils[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ram_wb[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in stream_utils-1.0[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in elf-loader[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_sdram_ctrl[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_altera_ddr_wrapper[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_intercon[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_intercon-1.0[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in fifo-1.0[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in vlog_tb_utils[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ram_wb[0m
[1;33mWARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in stream_utils-1.0[0m
[1;37mINFO:  Preparing jtag_tap[0m
[1;37mINFO:  Downloading olofk/jtag from github[0m
[1;37mINFO:  Preparing verilog-arbiter[0m
[1;37mINFO:  Downloading bmartini/verilog-arbiter from github[0m
[1;37mINFO:  Preparing verilog_utils[0m
[1;37mINFO:  Preparing vlog_tb_utils-1.0[0m
[1;37mINFO:  Preparing wb_bfm[0m
[1;37mINFO:  Preparing wb_intercon[0m
[1;37mINFO:  Preparing wb_common[0m
[1;37mINFO:  Preparing wb_ram-1.0[0m
[1;37mINFO:  Preparing wb_avalon_bridge[0m
[1;37mINFO:  Preparing adv_debug_sys[0m
[1;37mINFO:  Downloading olofk/adv_debug_sys from github[0m
[1;37mINFO:  Preparing altera_virtual_jtag[0m
[1;37mINFO:  Preparing mor1kx[0m
[1;37mINFO:  Downloading openrisc/mor1kx from github[0m
[1;37mINFO:  Preparing uart16550-1.5[0m
[1;37mINFO:  Downloading olofk/uart16550 from github[0m
[1;37mINFO:  Preparing elf-loader[0m
[1;37mINFO:  Preparing gpio[0m
[1;37mINFO:  Preparing vga_lcd[0m
[1;37mINFO:  Downloading olofk/vga_lcd from github[0m
[1;37mINFO:  Preparing i2c[0m
[1;37mINFO:  Downloading olofk/i2c from github[0m
[1;37mINFO:  Preparing sockit[0m

[1;31mERROR: Failed to build FPGA: "make" exited with an error code.
ERROR: See stderr for details.[0m
[Pipeline] sh
[orpsoc-cores] Running shell script
+ mkdir system-sockit
+ mv fusesoc.log system-sockit
[Pipeline] step
Archiving artifacts
[Pipeline] }
[Pipeline] // node
[Pipeline] }
[Pipeline] // parallel
[Pipeline] End of Pipeline
Finished: SUCCESS
```

oleg-nenashev, 2016-09-08:
It happens in http://ec2-52-209-86-82.eu-west-1.compute.amazonaws.com/job/playground/job/orpsoc-cores/10/

imphil, 2017-03-23:
PR #146  fixes this

olofk, 2017-03-29:
PR #146 is pushed now. Please close if this fixes the issue

imphil, 2017-07-20:
@olof that's fixed for sure, you can close it.

closed 2017-07-20 by olofk