olofk/fusesoc #122: Special backend: vivado-yosys

wallento, 2016-10-04:
It would be great to launch a variant where the synthesis is done by yosys and only the implementation by vivado. If I understand it correctly, Yosys can:

 - Compile to EDIF or RTL Verilog with instantiated Xilinx primitives for 7-Series FPGA
 - Compile to EDIF or RTL Verilog for others

olofk, 2016-10-04:
The plan is to add an option to select which synthesis tool to use, and take in the resulting EDF tool. Another use case is for people using Synplify for synthesis

wallento, 2016-10-04:
Yeah, I generally agree, but I think it is possible to integrate it even in the TCL project flow in vivado (only run when files changed). This is probably only a thing for developers, but can turn out pretty useful. That's also why I would stick to the synplify flow, that calls xilinx tools on its own.

olofk, 2018-11-27:
I implement support for using Yosys synthesis with Vivado here https://github.com/olofk/edalize/tree/yosys-vivado. I don't have any hw to test this with, so I'm waiting until someone can verify it works

imphil, 2020-12-18:
We do have that now in edalize.

closed 2020-12-18 by imphil