olofk/fusesoc #124: Cannot compile simulate de0_nano using Icarus?

trimcao, 2016-10-30:
Do I have to use modelsim for de0_nano simulation?

command: ~/sim$ fusesoc sim de0_nano --vcd --timeout=100000 --bootrom_file=spi_uimage_loader.vh 

Output: 
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_altera_ddr_wrapper
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in stream_utils-1.0
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in vlog_tb_utils
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in elf-loader
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_sdram_ctrl
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ram_wb
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in fifo-1.0
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_intercon
WARN:  plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in wb_intercon-1.0
INFO:  Preparing elf-loader
INFO:  Preparing jtag_vpi-r2
INFO:  Preparing mt48lc16m16a2
INFO:  Preparing s25fl064p-1.7
INFO:  Preparing vlog_tb_utils-1.0
INFO:  Preparing adv_debug_sys
INFO:  Preparing altera_virtual_jtag
INFO:  Preparing gpio
INFO:  Preparing i2c-1.13
INFO:  Preparing jtag_tap-1.13
INFO:  Preparing mor1kx-3.1
INFO:  Preparing or1k_bootloaders-0.9
INFO:  Preparing simple_spi-1.6
INFO:  Preparing uart16550-1.5.4
INFO:  Preparing wb_bfm-1.0
INFO:  Preparing verilog-arbiter-r1
INFO:  Preparing verilog_utils
INFO:  Preparing wb_common
INFO:  Preparing wb_intercon-1.0
INFO:  Preparing wb_ram-1.0
INFO:  Preparing wb_sdram_ctrl-r2
INFO:  Preparing wiredelay
INFO:  Preparing de0_nano

INFO:  Running /home/esc/.local/share/orpsoc-cores/cores/elf-loader/check_libelf.sh
INFO:  Running /home/esc/.local/share/orpsoc-cores/systems/de0_nano/legacy_symlink.sh
/home/esc/Desktop/csaw_esc_2016/sim/build/de0_nano/src/de0_nano
/home/esc/Desktop/csaw_esc_2016/sim/build/de0_nano/src/de0_nano_0
Compiling /home/esc/Desktop/csaw_esc_2016/sim/build/de0_nano/src/elf-loader/elf-loader.c...
Compiling /home/esc/Desktop/csaw_esc_2016/sim/build/de0_nano/src/elf-loader/vpi_wrapper.c...
Making elf-loader.vpi from  elf-loader.o vpi_wrapper.o...
Compiling /home/esc/Desktop/csaw_esc_2016/sim/build/de0_nano/src/jtag_vpi-r2/jtag_vpi.c...
Making jtag_vpi-r2.vpi from  jtag_vpi.o...
iverilog: invalid option -- 'P'
ERROR: Failed to build simulation model
ERROR: Failed to compile Icarus Simulation model

trimcao, 2016-10-31:
I also tried modelsim and it still did not work :(

> INFO:  Running /home/esc/.local/share/orpsoc-cores/cores/elf-loader/check_libelf.sh
> INFO:  Running /home/esc/.local/share/orpsoc-cores/systems/de0_nano/legacy_symlink.sh
> /home/esc/Desktop/csaw_esc_2016/sim/build/de0_nano/src/de0_nano
> /home/esc/Desktop/csaw_esc_2016/sim/build/de0_nano/src/de0_nano_0
> ld: cannot find -lelf
> ERROR: Failed to build simulation model
> ERROR: Linking of elf-loader failed

I used Ubuntu 16.04 and ModelSim Altera 15.1

olofk, 2016-11-01:
Hi,

I'm not sure why you get the first error with Icarus. Perhaps your Icarus version is too old. Can you show me the output from the same command, but adding --verbose after fusesoc

For the other problem, it seems like it fails to find libelf. Make sure that the libelf headers are installed. I think they are in the libelf-dev package for ubuntu

olofk, 2016-11-25:
@trimcao Did you make any progress on this?

trimcao, 2016-11-27:
I could not do it, but I was done with this project weeks ago. Thanks anyway.

closed 2016-11-26 by trimcao