olofk/fusesoc #125: Initialization/Installation Problem

pfliegster, 2016-11-02:
Hello,

Running under Ubuntu 16.04LTS, Python 2.7.12, pip 9.0.0; I'm having a problem running the "fusesoc init" command in the installation instructions. I suspect that something got screwed up during installation. Here are the errors I receive:
```
VirtualBox:~/fusesoc$ fusesoc init
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2940, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 637, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 650, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 834, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (attrs 16.2.0 (/usr/local/lib/python2.7/dist-packages), Requirement.parse('attrs<16.1.0'), set(['okonomiyaki']))
VirtualBox:~/fusesoc$
```
In fact, running any of the fusesoc commands seem to fail similarly, so I think something went wrong during my install or update (the output from the **sudo pip install -e .** phase says I have fusesoc 1.5).

Another note: when running the "**sudo pip** ..." commands, it kept giving me an error about permissions and suggested I use the "**sudo -H** ..." -H flag option. Which I did, so perhaps something got out of whack here.
Any help correcting my install is greatly appreciated :-)

olofk, 2016-11-02:
Actually, this isn't your fault, or really my fault either :)

It's the stupid pip package manager that fails to install the correct version of one of the dependencies. You will hopefully get around this by running `sudo pip install -I attrs==16.0.0`. There is some more info [here](https://github.com/enthought/okonomiyaki/issues/243)

Let me know if that works

pfliegster, 2016-11-02:
Excellent! Thank you very much Olof! That fixed my problem.

closed 2016-11-02 by pfliegster