olofk/fusesoc #126: fusesoc sim de0_nano fails

berndca, 2016-11-22:
I'm trying to simulate de0_nano using the icarus simulator. I'm getting the following messages:

INFO:  Running /home/bernd/.local/share/orpsoc-cores/cores/elf-loader/check_libelf.sh
INFO:  Running /home/bernd/.local/share/orpsoc-cores/systems/de0_nano/legacy_symlink.sh
/home/bernd/PycharmProjects/mix/workspace/de0_nano/build/de0_nano/src/de0_nano
/home/bernd/PycharmProjects/mix/workspace/de0_nano/build/de0_nano/src/de0_nano_0
Compiling /home/bernd/PycharmProjects/mix/workspace/de0_nano/build/de0_nano/src/elf-loader/elf-loader.c...
Compiling /home/bernd/PycharmProjects/mix/workspace/de0_nano/build/de0_nano/src/elf-loader/vpi_wrapper.c...
Making elf-loader.vpi from  elf-loader.o vpi_wrapper.o...
Compiling /home/bernd/PycharmProjects/mix/workspace/de0_nano/build/de0_nano/src/jtag_vpi-r2/jtag_vpi.c...
Making jtag_vpi-r2.vpi from  jtag_vpi.o...
warning: Found both default and `timescale based delays. Use
         -Wtimescale to find the module(s) with no `timescale.
../src/s25fl064p-1.7/s25fl064p.v:729: error: Unable to bind wire/reg/memory `tdevice_DP' in `orpsoc_tb.spi_flash.TDPr'
../src/s25fl064p-1.7/s25fl064p.v:738: error: Unable to bind wire/reg/memory `tdevice_RES' in `orpsoc_tb.spi_flash.TRESr'
../src/s25fl064p-1.7/s25fl064p.v:1352: error: Unable to bind wire/reg/memory `tdevice_PP' in `orpsoc_tb.spi_flash.pdone_process'
../src/s25fl064p-1.7/s25fl064p.v:1374: error: Unable to bind wire/reg/memory `tdevice_WR' in `orpsoc_tb.spi_flash.wdone_process'
../src/s25fl064p-1.7/s25fl064p.v:1390: error: Unable to bind wire/reg/memory `tdevice_BE' in `orpsoc_tb.spi_flash.erase'
../src/s25fl064p-1.7/s25fl064p.v:1394: error: Unable to bind wire/reg/memory `tdevice_PE' in `orpsoc_tb.spi_flash.erase'
../src/s25fl064p-1.7/s25fl064p.v:1398: error: Unable to bind wire/reg/memory `tdevice_SE' in `orpsoc_tb.spi_flash.erase'
7 error(s) during elaboration.
ERROR: Failed to build simulation model
ERROR: Failed to compile Icarus Simulation model

Please advise.
Thanks, Bernd

olofk, 2016-11-22:
Not entirely sure, but I've seen people reporting similar problems quite recently. I think it could be that your Icarus version is too old. Which version are you using? Could you try to update it, or run with modelsim (by supplying --sim=modelsim after the sim command)

berndca, 2016-11-22:
Good point. Failed with iverilog v0.9.7 installed with apt-get. Worked with v10.1 installed from github repo.

closed 2016-11-22 by berndca