olofk/fusesoc #128: Make fusesoc more decentralized

cactorium, 2016-12-01:
Sorry if this is the wrong channel for this kind of request; would it be possible to design fusesoc to be more flexible in terms of where it sources repositories? Currently the only way to hack (AFAIK) on some of the components is to maintain a fork of orpsoc-cores and modify the system and core files there. Would it be possible to allow fusesoc to source system and core files from other repositories? So instead of a single repository holding all the dependency and build information, it could keep using that one repository, but users can create their own systems and cores in separate repositories that they can source from as well. This seems really rigid compared to the package management and build systems in a lot of other languages, where you can basically source packages from arbitrary repositories if you give the package manager the address of it

Sorry if this is already possible, and thank you for your time!

olofk, 2016-12-01:
Hi,

Yes, this is already possible, but since the documentation is almost non-existent, I'm not surprised you could not find it. I actually took this as an opportunity to write a bit more documentation, so check out https://github.com/olofk/fusesoc/blob/master/doc/fusesoc.adoc#core-search-order where this is described now.

So no more forks of orpsoc-cores. Instead, do one of the following.
If it's a local core (i.e. no [provider] section), copy the directory of the core somewhere else and add the new locations as a core library. Either add the new location to your fusesoc.conf or specify it by running `fusesoc --cores-root=/path/to/directory ...`. You can verify that FuseSoC has picked up the new location by running `fusesoc core-info <your core>` and look at the Core root

If it's a remote core (i.e. with a [provider] section), you must first download the repository specified in the provider section. After that you can copy the .core file from orpsoc-cores into the downloaded repository and remove the [provider] section to avoid FuseSoC from trying to download it again. Some cores already have .core files in their repositories and in those cases, you don't need to copy the one from orpsoc-cores. You now have a local cores and can use the instructions outlined above. Hope this helps and let me know if I can add anything else to the documentation to make this clearer

cactorium, 2016-12-02:
Oh, that's exactly what I needed! Thank you for the documentation, it's clear enough for me at least

closed 2016-12-02 by cactorium