olofk/fusesoc #129: Development with cores from fusesoc

Fatsie, 2016-12-11:
In the readme it says fusesoc allows to create compile-time configurations. It is not clear to me how to do that.
Let's say that I want create a new project that uses the i2c core. How do I do that ? I tried to 'fusesoc fetch i2c' but this does not seem to give me the code.
Is this a use case not meant to be solved by fusesoc ?

olofk, 2016-12-11:
Hi,

I think this is best answered as three questions.

1. When I talk about compile-time configurations, I mean that some parameters of a core can be decided at compile-time. If you run for example `fusesoc build de0_nano --help` or `fusesoc sim wb_bfm --help`, you will see number of parameters (--bootrom_file, --spi_flash_file, --heartbeat, --tapfile etc..) that can be used to set some configuration for builds and simulations

2. `fusesoc fetch i2c` will download the i2c core to the FuseSoC cache, which by default is located in ~/.cache/fusesoc

3. If you want to create a project that uses the i2c core, you will create a new core that has a dependency on the i2c core. I recently added some documentation on how to get started with creating your own core [here](https://github.com/olofk/fusesoc/blob/master/doc/createcore.adoc). It's pretty basic right now, so please let me know if there is anything I should add to make it clearer.

Hope this help and let me know if you need any more assistance to get started

Fatsie, 2016-12-16:
OK understood. Will play and hopefully I find the time to document some things.

I know fusesoc is modeled after the packaging of programs for Linux etc. But there you typically have an upstream and downstream. Meaning that the packaging is maintained by other people than the one writing the software. Given limited resources I think for RTL cores it would be more opportune to make fusesoc more a natural member of the RTL cores development process.

closed 2016-12-16 by Fatsie

olofk, 2016-12-16:
I agree. I'd love to have more package maintainers and RTL developers writing their own .core files already when they start developing their cores. It's slowly happening, but I still write most .core files myself.