olofk/fusesoc #131: add svf-file generation

frantony, 2016-12-18:
@olofk 
Most of the time I use openocd to burn Altera's FPGA (e.g. I use Raspberry Pi for burning FPGA and can't run Quartus on it).
Can we add svf-file generation feature to fusesoc?
Just now I use this patch:
```
--- a/fusesoc/build/quartus.py
+++ b/fusesoc/build/quartus.py
@@ -8,7 +8,7 @@ class Quartus(Backend):
 
     MAKEFILE_TEMPLATE = """
 
-all: sta
+all: svf
 
 project: $(TCL_FILE)
        quartus_sh $(QUARTUS_OPTIONS) -t $(DESIGN_NAME).tcl
@@ -25,6 +25,9 @@ asm: fit
 sta: asm
        quartus_sta $(QUARTUS_OPTIONS) $(DESIGN_NAME)
 
+svf: sta
+       quartus_cpf -c -n p -q 2MHz -g 3.3 $(DESIGN_NAME).sof $(DESIGN_NAME).svf
+
 clean:
        rm -rf *.* db incremental_db
 """

```
I think that the patch is not perfect (e.g. hardcoding ``-q 2MHz -g 3.3 `` is a bad solution, I have to move this options to configuration file).

Any suggestions?

olofk, 2016-12-19:
Hi,

That's one way to solve your issue, but I'm always hesitating a bit to add more options directly into FuseSoC and prefer to keep these things in tcl files whenever possible. For example, I have a similar situation where I need to generate an rbf file with quartus. To do this, I add `set_global_assignment -name GENERATE_RBF_FILE ON` to one of the tcl files that is loaded from FuseSoC (to load a tcl file, add it to a fileset or create a new one, and set `usage=quartus` and `file_type=tclSource`). I haven't checked, but there is likely some way to do this. Another option would be to create a script that runs quartus_cpf and add it to the core file as a `post_impl_script` in the `[scripts]` section

frantony, 2016-12-20:
Thank you for your suggestion!

I have found ```GENERATE_CONFIG_SVF_FILE``` and ```GENERATE_SVF_FILE``` options in the [Quartus Prime Settings File Reference Manual](https://altera.com/content/dam/altera-www/global/en_US/pdfs/literature/manual/mnl_qsf_reference.pdf). I have enabled these options and svf-files were succesfully generated without any changes in the fusesoc codes.

closed 2016-12-20 by frantony