olofk/fusesoc #132: local provider

Fatsie, 2016-12-30:
In order to include .core files I propose to add a local provider that will look for files relative to the .core file and not download it. Proposed init section format

    [provider]
    name = local
    root = _rootpath_

\_rootpath_ is the relative path to the .core file location from which the other files form the filesets will be searched.
If this is implemented one can add the path to where one does RTL development to fusesoc.conf and from then on the cores will be found from the dev place there.
It should also be easy to write a release script that would then transform the .core file that later on can be pushed to fusesoc-cores.
Let me know what you think, if it sounds OK I may even try to implement it and provide pull request.

Background it that I am playing with some 8-bit cpus from freecores and I do plan to add .core in their code.

wallento, 2016-12-30:
Hi,

isn't that the default provider anyways?

If you create a fileset like this

    [fileset src_files]
    files = file1.v file2.v
    file_type = verilogSource

it looks for the files in the folder. Or do I understand it wrong?

Cheers,
Stefan

Fatsie, 2016-12-30:
Indeed. It was not clear to me from the documentation and I forgot how it was done in the tutorial.

closed 2016-12-30 by Fatsie