olofk/fusesoc #134: How to use coregen provider

Fatsie, 2017-01-04:
I am trying to use coregen from a core file using the coregen provider. This is the .core file:

    CAPI=1
    [main]
    name = freecores.github.io:t80:Z80Rom
    description = ROM for Z80 debug
    simulators = isim
    
    [provider coregen]
    script_file = syn/xilinx/coregen/Z80Rom.xco
    project_file = syn/xilinx/coregen/Atlys.cgp

For the t80 core I have the following depend in the [main] section:

    [main]
    ...
    depend = freecores.github.io:t80:Z80Rom

When trying to simulate the top core with isim the files are not generated:

    $ fusesoc sim --build-only freecores.github.io:t80:T80
    INFO:  Preparing freecores.github.io:t80:Z80Rom:0
    INFO:  Preparing freecores.github.io:t80:T80:0
    
    Running: /home/eda/software/Xilinx/14.7/ISE_DS/ISE/bin/lin64/unwrapped/fuse TestBench -prj isim.prj -o fusesoc.elf 
    ISim P.20131013 (signature 0xfbc00daa)
    Number of CPUs detected in this system: 4
    Turning on mult-threading, number of parallel sub-compilation jobs: 8 
    Determining compilation order of HDL files
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80_Pack.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/GBse.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T8080se.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80_ALU.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80a.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80_MCode.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80_Reg.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80sed.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80se.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80s.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T80.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/Z80.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/SSRAM.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/T16450.vhd" into library work
    Parsing VHDL file "../src/freecores.github.io_t80_T80_0/rtl/vhdl/DebugSystem.vhd" into library work
    ERROR:HDLCompiler:104 - "../src/freecores.github.io_t80_T80_0/rtl/vhdl/DebugSystem.vhd" Line 116: Cannot find <z80rom> in library <work>. Please ensure that the library was compiled, and that a library and a use clause are present in the VHDL file.
    ERROR:HDLCompiler:854 - "../src/freecores.github.io_t80_T80_0/rtl/vhdl/DebugSystem.vhd" Line 33: Unit <struct> ignored due to previous errors.
    VHDL file ../src/freecores.github.io_t80_T80_0/rtl/vhdl/DebugSystem.vhd ignored due to errors
    ERROR: Failed to build simulation model
    ERROR: Failed to compile Isim simulation model

DebugSystem.vhd is where the Z80Rom block is used

Fatsie, 2017-01-07:
Ok found first error, the following

    [provider coregen]

should be

    [provider]
    name = coregen

With this change I still does not seem to get the right combination of fileset and this provider to in the end be able to generate the files and be used in the parent core.

olofk, 2017-01-08:
Sorry for the late reply. You need to add the HDL files that are generated by coregen to a fileset in the z80rom core, as FuseSoC has no knowledge of this otherwise

Fatsie, 2017-01-08:
Can you be more specific in how to specify them or a working .core file of a RAM done with coregen ?
Either it just seems fusesoc copies the already generated files without running the coregen or it tries to run coregen but that errors out with syn/xilinx/ROM80.xco.
What is actually the file_type to use for specifying the .xco file ?

Fatsie, 2017-01-09:
I'm trying to provide the T80 as a github hosted .core. Depending on custom C++ code needing configuration of compiler + Makefiles etc. does fit in this. That's why I am trying to generate the ROM with coregen from out of FuseSoC. In the end I think we should have even a more generic ROM generator not (directly) depending on Xilinx specific tools.

I have indeed an empty directory for the coregen core in ~/.cache, I think this should be removed if coregen fails or better check if needed files are there and redo the coregen if not. Will see what I can do and also see if I can get the coregen core working now.

olofk, 2017-01-09:
Agree. FuseSoC should be smarter about this and not assume the core is working if the directory exists. Cache handling needs more love in several areas

Also agree that custom c++ is really unnecessary here, and using coregen is bad too, since it limits the number of simulators we can use. I propose that we use a verilog module here that can be fed with a parameter containing the initial memory contents. This is super easy and is the way I usually do this for other systems. The reason for using verilog here is that we can use $readmemh, and VHDL doesn't have a similar standard for prepopulating memories. A simplified version of https://github.com/olofk/or1k_bootloaders/blob/master/wb_bootrom.v would work

olofk, 2017-01-09:
Because even if you use coregen, you still have the problem of how to fill the contents of the ROM

Fatsie, 2017-01-09:
Will look at pure verilog but this assumes all synthesis flow do the proper thing with $readmemh, as I also want to target ASICs I'm not sure how valid this assumption is.
Anyway I provided pull request #135 for the path generation of coregen anyway and this issue can be closed for the moment.

closed 2017-01-09 by Fatsie