olofk/fusesoc #137: isim and VHDL-2008

Fatsie, 2017-01-16:
I'm using std.env from VHDL-2008 for using the finish procedure in a testbench.
In order for ghdl to have access to this library the source has to be defined as 2008. (#136 is already fix for ghdl)
Problem now is that for ISim in the project file vhdlSource-2008 are using vhdl2008 command. Problem is that my version of ISim does not recognize it:

    ERROR:Simulator:722 - First word "vhdl2008" is not one of keywords vhdl or verilog

I don't know if this is because I am using the Lite version from ISE 13.7 Webpack.
If I just use vhdl as for the other vhdl files ISim works and std.env library is also available.

olofk, 2017-01-17:
Ouch. That was a mistake. Isim doesn't support VHDL 2008 (even though your comment implies that it supports some part of it). This was likely a copy/paste error from xsim, which does support vhdl 2008.

I'll remove the keyword from isim, to give isim a chance to compile it as a regular vhdl file.

olofk, 2017-01-17:
Should be fixed in 1cb2012c33f7fe9e54c87b9a9dc91d70783f9da7

Fatsie, 2017-01-17:
Tested latest master and can run my simple T80 testbench with isim and ghdl from the same .core.

closed 2017-01-17 by Fatsie