olofk/fusesoc #143: "attrs" version conflict

Sawaiz, 2017-03-17:
On installing fusesoc (using windows subsystem for linux which is just ubuntu) via pip (`sudo pip install fusesoc`), or though git, and then running `fusesoc`, with or without any args, gives this error
```
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2749, in <module>
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 446, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 459, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 632, in resolve
    raise VersionConflict(dist,req) # XXX put more info here
pkg_resources.VersionConflict: (attrs 16.0.0 (/usr/local/lib/python2.7/dist-packages), Requirement.parse('attrs>=16.1.0'))
```
If I force the upgrade of attrs to 16.3.0 with  `sudo pip install --upgrade attrs` , it uninstalls attrs 16.0.0, and updates it to 16.3.0 which is verified with `pip show attrs`. Running `fusesoc` gives this error.
```
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2749, in <module>
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 446, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 459, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 628, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: attrs==16.0.0
```
Where it wants the lower version again. I have jump back and forth (`sudo pip install --upgrade attrs==16.0.0 `), and try other versions (16.0.0, 16.1.0, 16.2.0, 16.3.0) but it all yields the same errors. As I can't install both versions, is this a error in the package requirements?

Python 2.7
pip 1.5.4

olofk, 2017-03-17:
This is the stupid stupid pip package manager that can't properly handle dependencies, so I had to introduce a workaround. This workaround is now coming back to bite me, as simplesat (and its dependency okonomiyaki) has been updated. I'm working on fixing this now, and it has also been reported in https://github.com/openrisc/or1k-tests/issues/2

Please try the depfix branch and see if that fixes the issues. I want to push them to master as soon as I get confirmation that they fix the issue.

If that doesn't help, you could try to temporarily fix this is byt installing simplesat version 0.7.0 (`pip install simplesat==0.7.0` should hopefully work).

berndca, 2017-03-17:
You might want to consider using Armin Ronachers [pipsi](https://github.com/mitsuhiko/pipsi) script to install fusesoc. It automatically creates a virtualenv and installs fusesoc in this virtualenv thereby avoiding conflicts.

berndca, 2017-03-17:
It appears that the workaround for the attr package is no longer needed for simplesat 0.8.0 since both simplesat 0.8.0 and it's dependency okonomiyaki 0.16.6 both require attrs >= 0.16.1. Just requiring simplesat 0.8.0 should be sufficient. There is no need to require attrs directly from fusesoc.

It seems that the problems reported might be related to the fact that fusesoc was installed globally. This is pretty much always a bad idea. Always install into a virtualenv.

Sawaiz, 2017-03-17:
My first installations were with a completely stock python setup, so I don't think there were any dependency conflicts because nothing else was ever installed. Switching to the deepfix branch no longer gives me errors, and lets me install and run properly.

closed 2017-03-17 by Sawaiz