olofk/fusesoc #149: Fusesoc cannot download opencores dependencies

ajaymills, 2017-03-30:
Hi, I am sure this must have come up before. I am trying to build the system for Atlys.

The call to "opensoc" fails at the first dependency which is ac97:

```
INFO: Preparing ::ac97:0
INFO: Downloading ac97 from OpenCores
svn: E170013: Unable to connect to a repository at URL 'http://opencores.org/ocsvn/ac97/ac97/trunk/rtl/verilog'
svn: E000110: Error running context: Connection timed out
ERROR: "svn co -q --no-auth-cache -r 21 --username orpsoc --password orpsoc http://opencores.org/ocsvn/ac97/ac97/trunk/rtl/verilog /home/ajmills/.cache/fusesoc/ac97_0" exited with an error code.
ERROR: See stderr for details.
```

I understand that the OpenCores SVN has been flaky for a long time and was phased out.  I had thought that these components were migrated to here on Github. Is there some way to work around this?

olofk, 2017-04-01:
Yeah, as you say, the connection is flaky and most cores have been migrated to use a source from github instead, but there are still some remaining. I fixed ac97 now to fetch from github instead, so it should hopefully work after a `fusesoc update`. I took the opportunity to add support for the testbench too, so `fusesoc sim ac97` should run the regression tests. Let me know if there are more problems

olofk, 2017-04-11:
Closing this now. Feel free to reopen

closed 2017-04-11 by olofk