olofk/fusesoc #150: Meaningful error for extra whitespaces

wallento, 2017-04-21:
If the user accidentally adds a whitespace in a filelist between the filename and the attributes, the error (exception) is not very helpful.

Correct:

    [fileset dpi]
        file_type = cppSource
        usage = sim verilator
        files =
          dpi/myheader.h[is_include_file]

Incorrect:

    [fileset dpi]
        file_type = cppSource
        usage = sim verilator
        files =
          dpi/myheader.h [is_include_file]

In the incorrect case, fusesoc takes the whitespace as the separator and an empty filename with the attribute leading to the exception:

    IOError: [Errno 21] Is a directory: '/path/to/module/'

It would be more helpful to throw an error like "empty filename, did you add an extra space between the filename and the attribute?"

fastwalker2k4, 2017-05-09:
I tried to reproduce the bug by using 'tests.test_common.get_core' on the 'tests/cores/mor1kx-generic' file, since it contains filenames with attributes. It completes without changes, fails as expected after adding some junk, but unexpectedly completes after adding a space between filename and attribute (e.g. "user_file [file_type=user])".

This is quite literally a first bug for me, so it could be an error on my end. I could some input; does the issue still exist?

olofk, 2017-05-10:
The problem still exists, but I figured out why it wasn't triggered. The file you chose to change is of type user, which means that it is just exported, but no backends include it in their configuration files. Also, the test_common module does not contain test cases, but helper functions for the other tests (maybe there's a better way to structure this. I'm quite new to pytest).

With that said, you can reproduce the problem for example by inserting the space in `vlog_file.v[file_type=verilogSource]` in `tests/cores/misc/filetypes.core` and run test_ise

Let me know if this works

olofk, 2020-02-29:
Closing this as CAPI1 has been deprecated for quite some time

closed 2020-02-29 by olofk