olofk/fusesoc #159: Fusesoc Sockit build fails

Godtec, 2017-06-27:
After installing fusesoc and suscessfully building de0-nano.
I tried to build the sockit build, and this the following error I received.

Warning (10306): Invalid value "block_ram" for synthesis attribute "syn_ramstyle" at ../src/vga_lcd_0/rtl/verilog/generic_dpram.v(155) File: /home/mikek/Documents/Mike_Play_Learn/fusesoc/build/sockit_0/src/vga_lcd_0/rtl/verilog/generic_dpram.v Line: 155
Error (12002): Port "wb_rty_o" does not exist in macrofunction "ram_wb0" File: /home/mikek/Documents/Mike_Play_Learn/fusesoc/build/sockit_0/src/sockit_0/rtl/verilog/orpsoc_top.v Line: 930
Warning (12241): 76 hierarchies have connectivity warnings - see the Connectivity Checks report folder
Info (144001): Generated suppressed messages file /home/mikek/Documents/Mike_Play_Learn/fusesoc/build/sockit_0/bld-quartus/sockit_0.map.smsg
Error: Quartus Prime Analysis & Synthesis was unsuccessful. 1 error, 189 warnings
    Error: Peak virtual memory: 4474 megabytes
    Error: Processing ended: Tue Jun 27 17:05:02 2017
    Error: Elapsed time: 00:05:58
    Error: Total CPU time (on all processors): 00:06:16
make: *** [map] Error 3
ERROR: Failed to build FPGA: "make" exited with an error code.
ERROR: See stderr for details.

Any idea's how to solve, thanks.

MikeK

Godtec, 2017-06-28:
Failing this, I opened the project in Quartus, I then removed the pin from the project and recompiled.
It worked, I have the Sockit board, and testing it next.

Will post on outcome.

Thanks,
MikeK

olofk, 2017-06-28:
Hmm... I changed the sockit port to use the wb_ram component instead of the deprecated ram_wb component a while ago. Could it be that your base library (orpsoc-cores) is out of date. Could you try to run `fusesoc update` to see if that fixes the problem?

Godtec, 2017-06-28:
Hi Olof:

Hi Did exactly that, fusesoc update, and same issue.

mikek@mike-M6700:~/Documents/Mike_Play_Learn/fusesoc$ fusesoc update
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::elf-loader:0
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::fifo:1.0
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::ram_wb:0
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::stream_utils:1.0
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::vlog_tb_utils:0
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::wb_altera_ddr_wrapper:0
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::wb_intercon:1.0
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::wb_intercon:0
WARNING: plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters in ::wb_sdram_ctrl:0
INFO: Updating '/home/mikek/.local/share/fusesoc/orpsoc-cores'
Already up-to-date.
INFO: Updating '/home/mikek/.local/share/fusesoc/fusesoc-cores'
Already up-to-date.
mikek@mike-M6700:~/Documents/Mike_Play_Learn/fusesoc$ 

And here is the same error again.

Warning (10306): Invalid value "block_ram" for synthesis attribute "syn_ramstyle" at ../src/vga_lcd_0/rtl/verilog/generic_dpram.v(155) File: /home/mikek/Documents/Mike_Play_Learn/fusesoc/build/sockit_0/src/vga_lcd_0/rtl/verilog/generic_dpram.v Line: 155
Error (12002): Port "wb_rty_o" does not exist in macrofunction "ram_wb0" File: /home/mikek/Documents/Mike_Play_Learn/fusesoc/build/sockit_0/src/sockit_0/rtl/verilog/orpsoc_top.v Line: 930
Warning (12241): 76 hierarchies have connectivity warnings - see the Connectivity Checks report folder
Info (144001): Generated suppressed messages file /home/mikek/Documents/Mike_Play_Learn/fusesoc/build/sockit_0/bld-quartus/sockit_0.map.smsg
Error: Quartus Prime Analysis & Synthesis was unsuccessful. 1 error, 189 warnings
    Error: Peak virtual memory: 4474 megabytes
    Error: Processing ended: Wed Jun 28 18:48:11 2017
    Error: Elapsed time: 00:08:53
    Error: Total CPU time (on all processors): 00:09:00
make: *** [map] Error 3
ERROR: Failed to build FPGA: "make" exited with an error code.
ERROR: See stderr for details.
mikek@mike-M6700:~/Documents/Mike_Play_Learn/fusesoc$ 

Thanks.
MikeK

olofk, 2017-06-29:
There was a very simple explanation for this. I had fixed this locally, but never commited the change. It's done now, so hopefully it should work after a `fusesoc update` now. Please let me know if this helps

Godtec, 2017-06-29:
Yup!

Works now!

Great Work!
Thanks For all your Help!
MikeK

olofk, 2017-06-29:
Happy to hear it works. Unfortunately I no longer have a sockit board myself, so I can't check that the actual image works. Please file a bug against https://github.com/openrisc/orpsoc-cores if there are any runtime issues

closed 2017-06-29 by olofk