olofk/fusesoc #160: Windows PATH problem

bqwer, 2017-07-02:
Hello.
I have following issue: when I try to build system I get message `ERROR: Failed to build FPGA: Command 'vivado' not found. Make sure it is in $PATH`
Strange part is that I have $PATH set up properly. I can run `vivado` from cygwin, cmd and powershell without any additional steps.
I use fusesoc 1.6.1 installed via pip, python 3.6, Windows 10. I could not install fusesoc from git repo on windows as `pip install -e .` fails.
Please help solve this.

bqwer, 2017-07-02:
Well, I think I found cause in fusesoc/utils.py
Please check answers here (not just first):
https://stackoverflow.com/questions/3022013/windows-cant-find-the-file-on-subprocess-call

olofk, 2017-07-03:
Thanks for reporting and finding a potential fix. ok, so if I understand it correctly, all I need to do is set shell=True for windows platforms. Would you be able to test that? Unfortunately I don't have a windows environment here.

I will need to look into why installing from git doesn't work too. Could you file a separate bug for that so I can easier keep track on that

bqwer, 2017-07-03:
Yes, setting `shell=True` in `fusesoc/utils.py` fixed the problem. But then I got plenty of other with paths in windows environment (i'm guessing `\` and `/`) and could not build my project. I will try to post this issues later.

olofk, 2017-07-03:
Python automatically uses \ instead of / on windows platforms, but I have seen that both modelsim and quartus actually uses forward slashes on windows too. After some googling it seems like I should do the same for vivado. It might help with this (completely untested) patch

    diff --git a/fusesoc/build/vivado.py b/fusesoc/build/vivado.py
    index 81bcdcc..0182f4c 100644
    --- a/fusesoc/build/vivado.py
    +++ b/fusesoc/build/vivado.py
    @@ -100,15 +100,15 @@ class Vivado(Backend):
                 design       = self.name,
                 part         = self.tool_options['part'],
                 bitstream    = self.name+'.bit',
    -            incdirs      = ' '.join(incdirs),
    +            incdirs      = ' '.join([i.replace('\\', '/') for i in incdirs]),
                 ip           = ipconfig,
                 parameters   = parameters,
                 extras       = extras,
    -            tcl          = '\n'.join(['source '+s for s in tcl]),
    -            src_files    = '\n'.join(['read_verilog '+s for s in verilog]+
    -                                     ['read_verilog -sv '+s for s in sverilog]+
    -                                     ['read_vhdl '+s for s in vhdl]),
    -            xdc_files    = '\n'.join(['read_xdc '+s for s in constr])))
    +            tcl          = '\n'.join(['source '+s.replace('\\', '/') for s in tcl]),
    +            src_files    = '\n'.join(['read_verilog '+s.replace('\\', '/') for s in verilog]+
    +                                     ['read_verilog -sv '+s.replace('\\', '/') for s in sverilog]+
    +                                     ['read_vhdl '+s.replace('\\', '/') for s in vhdl]),
    +            xdc_files    = '\n'.join(['read_xdc '+s.replace('\\', '/') for s in constr])))
     
             tcl_file.close()

Paebbels, 2017-07-03:
That's because it's TCL code. TCL is allergic to `\`...

olofk, 2017-07-04:
I'm usually allergic to TCL, but in this case I think TCL and I agree :)

olofk, 2017-07-10:
I got the opportunity to try this on a windows computer now, and have pushed some patches that seem to work for me. Please close if you can confirm it has been fixed

bqwer, 2017-07-13:
Yes it works now! At least I was able to install it and build my test project.
Here are several tips:
1. You obviously need python to be installed. I used version 3.6
2. I couldn't use fusesoc from cygwin
3. You must be able to run `git` from windows shell of your choise (cmd/powershell) before installing fusesoc. I had success with https://git-for-windows.github.io/
4. When installing fusesoc you should run shell with admin privileges 
5. You_may_ need to update pbr and setuptools using `pip install -U pbr setuptools_scm` command
6. Make sure that all your building tools are in widows `PATH` environment variable

With this I was able to install fusesoc from git using `pip install -e .` using powershell and perform `build` command. For me it's really good news as now I can show fusesoc to my students in class with windows PCs.

olofk, 2017-08-05:
Happy to hear that it's working now. Please let me know if you come across other issues, as I want your students to have a good first experience. Thanks for reporting and for the detailed instructions

closed 2017-08-05 by olofk