olofk/fusesoc #161: Override files from another core

lucasax, 2017-07-28:
I'm creating a slightly modified version of the or1200 processor with an additional pipeline stage, and  most of the files that I'm using are from the original core. I want to move my modified files to a separate core in order to create a cleaner environment, so that I can include/exclude them.

What is the best practice to override files from another core?

olofk, 2017-07-28:
This is quite easy, although not documented unfortunately. If there exists a file in the directory of the .core file, this will be used instead of the one from the original repo. So these steps should hopefully work
1. Copy the original or1200 core file to a new directory
2. Add the path to this directory at the end of cores_root in fusesoc.conf (or add --cores-root=/path/to/or1200 to the command-line)
3. Add your new/modified files to the new directory with the correct subdirectories (e.g. rtl/verilog/newfile.v)
4. Add the new files to the .core file

Done! Out of curiousity, is there a reason why you want to extend the old or1200? Generally we recommend using [mor1kx](https://github.com/openrisc/mor1kx/) as that is better maintained and has more features

Let me know how this works out

olofk, 2017-07-28:
Also, as a general tip, you can run `fusesoc core-info or1200` and check the CORE ROOT line to verify that it has picked up your new version of the .core file

lucasax, 2017-07-28:
Amazing! Thank you for the clear answer and the great advice. 
I'm using this processor for my master's thesis. I've decided to use the or1200 instead of the mor1k because I've estimated that it requires the least amount of alterations of the original verilog description (at least for my goal).

olofk, 2017-07-29:
Glad to be of help. If you think that the outcome of your thesis would be useful to the wider open source silicon community, you can register the project at http://librecores.org/ to make it easier for other people to find it. Also, if you're interested in open source silicon in general, feel free to come to our yearly conference http://orconf.org/ in September.

Feel free to close the issue or let me know if you need more help

lucasax, 2017-08-29:
I've noticed that this technique doesn't work using verilator.

olofk, 2017-08-29:
Hmm... that is unexpected. I need to figure out why and construct a test case. It would help if have the possibility to give me code that shows this situation

lucasax, 2017-08-29:
After some testing I've realized that is probably my mistake. My project has a verilog include file that I've added under "src_files" with "[is_include_file]". This works fine with Modelsim but not with Verilator. Adding the same file under "include_files" solves the issue with both.

olofk, 2017-08-30:
Ah ok. Great that it works. If you need more flexibility with file types, I also suggest that you migrate from the src_files/include_files sections to filesets as outlined here https://github.com/olofk/fusesoc/blob/master/doc/migrations.adoc#migrating-to-filesets

Closing bug now

closed 2017-08-30 by olofk