olofk/fusesoc #166: Use templating engine for generated files

imphil, 2017-08-22:
Currently we put the templates of the generated project files (e.g. the TCL file for Vivado) inside the backend code itself. That's hard to maintain and annoying to modify.

How about using a templating engine like Jinja for that instead? I'll come up with a patch if you don't have fundamental issues with that @olofk.

In the long run, this would also enable us to let users specify their own template, which is then used and filled by fusesoc. [However, that's optional future stuff, nothing I'm thinking of implementing right now.]

olofk, 2017-08-22:
Great! jinja has been suggested before for something else (can't remember what right now). Happy to review such a patch. Also, the makefiles could use that

wallento, 2017-08-22:
+1 for jinja

benreynwar, 2017-08-22:
Yeah I really like jinja, and have found that the syntax works pretty
nicely with both Verilog and VHDL.

commit 74e96a22 (2018-05-02): Introduce Jinja2 for templating

Currently many backends have templates for the files they write out
(project files, TCL files, Makefiles, etc) included in the Python source
code of the backend. Other backends concatenate strings to create the
resulting output files.

This commit introduces Jinja2 to fusesoc, making it possible to use
Jinja2 templates instead.

No backends are converted in this commit, this will follow later.

Fixes #166
closed 2018-05-02 by olofk