olofk/fusesoc #167: Backends must not reorder source files

imphil, 2017-08-22:
Currently the Vivado backends (and others as well from a quick look) don't place HDL sources into the project file in the order they are resolved from the dependency resolver, but reorder the files. The Vivado backend, for example, groups the HDL files by file type (first all verilog files, then all sv files, then all vhdl files).

For most Verilog and VHDL files, the compile order doesn't matter.  However, sometimes it does matter -- for example, for package definitions in SystemVerilog. The SV standard requires that packages are defined before they are used, hence these files must be compiled first.

Note: Vivado doesn't seem to care about that rule and still compiles the resulting project without errors. However, SpyGlass isn't as lenient and fails the lint process. However, let's make sure we don't rely on lenient behavior by tools but do the right thing (TM).

TODOs:
- [ ] create testcases which check this behavior
- [ ] fix all backends showing to be problematic. Vivado is one of them.

olofk, 2017-08-22:
Except for the vivado backend, I'm pretty sure that I have been careful to have them in order. Both as they appear in the core file, and between cores, sorted as a flattened dependency tree. Vivado should be fixed, and if you find other places, that would be good to take care of too

imphil, 2018-05-02:
Fixed together with using Jinja2 as templating language for Vivado.

closed 2018-05-02 by imphil