olofk/fusesoc #169: Source generation sctipts

bqwer, 2017-09-03:
Hello. 
I have the question about adding sources which are generated with user scripts.

We have several cores where sources are generated from templates using our scripts (python).
Generated sources are mainly interconnects and  top level sources where many cores are connected together with many buses.
 
As far as I know this is not such uncommon situation.
I've searched issues, doc and source and I couldn't find way do it.
Was #75 about it? But I couldn't find  `[generate]` section handlers in source.
I've tried but couldn't find pre-export hook to call generation scripts (in that case maybe I could specify generated files in `.core` beforehand to include them on export?).

Is there any way to launch user generator and export resulting files with fusesoc?

olofk, 2017-09-03:
Hi,

Yes, this a weak point of FuseSoC. The generators are supposed to fix this, but as you have noticed, it is not implemented (there have been [some progress](https://github.com/benreynwar/fusesoc_generators) by @benreynwar lately, but it's not a general solution yet).

As you say, this is a pretty common situation. There are some cases where the scripts functionality can help out. As an example, I'm using a script to copy some extra files and generate a version register based on git tags and sha of the top-level source repository in one of my projects. The trick is to create dummy files that FuseSoC will export and then let a pre_build_script overwrite those files. This only works however if you know beforehand what the names of the files will be.

To your help you have some environment variables that can be used in the scripts
`WORK_ROOT` points to the working directory of the EDA tool (e.g. build/top_level_core/bld-tool) and `FILES_ROOT` points to the (exported) source directory of each core. There are also `BUILD_ROOT` and `SYSTEM_ROOT`, but those are only kept for backwards compatibility and will be removed at some point.

This also means that there is another limitation. Unless you run with the just recently added `--no-export` flag, there is no way to reference script files from the original repo locations. The solution here is to also list the script files in your .core file and use `file_type=user` for them. That way they will be exported, and you can reference the from the scripts section with `$FILES_ROOT/your_script.sh`

Hope this helps until the generators support is in place

wallento, 2018-01-13:
Hi, I have opened an RFC about templates here: #187. Would that cover your use case, @bqwer?

olofk, 2018-10-08:
This is hopefully fixed in FuseSoC master now

closed 2018-10-08 by olofk