olofk/fusesoc #170: Allow Chain Files to Program

addisonElliott, 2017-09-21:
I am working adding a DE10 Nano SoC system to FuseSoC. 

I run the following command to program the DE10 Nano:
`fusesoc pgm de10_nano`
and I receive the following error:
> Error (209015): Can't configure device. Expected JTAG ID code 0x02D020DD for device 1, but found JTAG ID code 0x4BA00477. Make sure the location of the target device on the circuit board matches the device's location in the device chain in the Chain Description File (.cdf).
> Error (209012): Operation failed

This is the result of running jtagconfig:
```
addison@ubuntu:~/Desktop/tutorials/de0_nano/build/de10_nano_0/bld-quartus$ jtagconfig
1) DE-SoC [1-1]
  4BA00477   SOCVHPS
  02D020DD   5CSEBA6(.|ES)/5CSEMA6/..
```

Fiddling around with it, I created a custom Chain file (CDF) to be able to program it. Here is the contents of the Chain file:
```
/* Quartus Prime Version 17.0.0 Build 595 04/25/2017 SJ Lite Edition */
JedecChain;
        FileRevision(JESD32A);
        DefaultMfr(6E);

        P ActionCode(Ign)
                Device PartName(SOCVHPS) MfrSpec(OpMask(0));
        P ActionCode(Cfg)
                Device PartName(5CSEBA6U23) Path("/home/addison/Desktop/tutorials/de0_nano/build/de10_nano_0/bld-quartus/") File("de10_nano_0.sof") MfrSpec(OpMask(1));

ChainEnd;

AlteraBegin;
        ChainType(JTAG);
AlteraEnd;
~           
```

I think it would be worthwhile to allow a Chain description file (CDF) to be specified in the *.system file. This would simply be copied to the bld-quartus directory. Afterwards, when the user calls `fusesoc pgm de10_nano`, it would search for the Chain file first and use that. Otherwise, it defaults to the regular quartus_pgm command.

Thoughts?

olofk, 2017-09-21:
Thanks for the detailed description. I think that could be a good improvement. Just one thing to get it out of the way, [please move away from .system files](https://github.com/olofk/fusesoc/blob/master/doc/migrations.adoc#migrating-from-system-files)

With that said, I need to think through the details. From your example there is a hardcoded path, which is probably best if we could avoid. I'm also a bit rusty on the Quartus programming stuff. How do you use the cdf file from the command line? Looking at `quartus_pgm -h`, there's also a possibility to specify an argument file with `-f`. Could that be an option instead, to let FuseSoC read an argument file and the user can specify all options inside of that?

addisonElliott, 2017-09-21:
So, this is the command I use for the custom chain file I've made:
` quartus_pgm --mode=jtag Chain.cdf`

It is pretty simple to use. However, the thing to note is that the chain file contains the path to the SOF file. As long as the chain file is in the same directory as the SOF file, then it should be easy enough to link to the SOF relatively.

I was unaware about .system files being deprecated. We're actually trying to port the OpenRISC to a DE10 Nano SoC in a graduate level class I'm in. If time permits, we may try to move away from the .system files. Thank you for letting me know about that

olofk, 2017-10-04:
Sorry for the late reply

Ah ok, I understand. You convinced me that it does make sense to make the cdf system-specific and use it if it's defined. Got a bit too much on my plate unfortunately, so can't work on this right now. Happy to review a patch though

imphil, 2020-12-18:
Not sure if that's relevant still, and probably an edalize issue either way. Please re-open if that problem comes up again.

closed 2020-12-18 by imphil

somhi, 2022-01-07:
I'm trying to add a Cyclone V board into fusesoc. I have the same problem as descrived above. It was finally resolved?

Info (209060): Started Programmer operation at Fri Jan  7 01:40:22 2022
Error (209015): Can't configure device. Expected JTAG ID code 0x02D020DD for device 1, but found JTAG ID code 0x4BA00477. Make sure the location of the target device on the circuit board matches the device's location in the device chain in the Chain Description File (.cdf).
ERROR: Failed to run fusesoc:utils:blinky:1.1 : '['quartus_pgm', '--mode=jtag', '-o', 'p;fusesoc_utils_blinky_1_1.sof']' exited with an error: 3

$ jtagconfig
1) Arrow 96 CV SoC Board [1-4]
  4BA00477   SOCVHPS
  02D020DD   5CSEBA6(.|ES)/5CSEMA6/..

somhi, 2022-01-09:
This works, but needs to be added somehow into fusesoc to define the FPGA is the second device

```
quartus_pgm --mode=jtag -o "p;fusesoc_utils_blinky_1_1.sof@2"
            # The @2 selects the second device in the JTAG chain, because this chip
            # puts the ARM cores first.
```