olofk/fusesoc #176: Don't throw and exception when encountering an unknown section in a core file

imphil, 2017-10-09:
From previous testings, I have a core file with a [spyglass] section. fusesoc (master as of today) recognizes this section as unsupported, but still throws an exception. Either the warning should be a fatal error, or (preferably) fusesoc should really ignore unknown sections.

~~~
WARNING: Unknown section 'spyglass' in '/home/no56hud/src/optimsoc/examples/fpga/vcu108/compute_tile/compute_tile_vcu108-spyglass.core'
Traceback (most recent call last):
  File "/home/no56hud/.local/bin/fusesoc", line 9, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/data/home/no56hud/src/fusesoc/fusesoc/main.py", line 413, in main
    run(parsed_args)
  File "/data/home/no56hud/src/fusesoc/fusesoc/main.py", line 313, in run
    cm.add_cores_root(cores_root)
  File "/data/home/no56hud/src/fusesoc/fusesoc/coremanager.py", line 166, in add_cores_root
    self.load_cores(os.path.expanduser(p))
  File "/data/home/no56hud/src/fusesoc/fusesoc/coremanager.py", line 152, in load_cores
    self.load_core(os.path.join(root, f))
  File "/data/home/no56hud/src/fusesoc/fusesoc/coremanager.py", line 135, in load_core
    core = Core(file)
  File "/data/home/no56hud/src/fusesoc/fusesoc/core.py", line 98, in __init__
    self.backend = getattr(self, self.main.backend)
AttributeError: 'Core' object has no attribute 'spyglass'
~~~

olofk, 2017-10-09:
FuseSoC warns about unknown sections, but carry on anyway. Without seeing the core file, I think the problem here is that the backend parameter in the main section is set to spyglass

I agree however that FuseSoC should catch this error and exit gracefully

olofk, 2017-10-10:
I added a patch now that ignores cores with an invalid backend parameter in the main section. I hoped to just ignore invalid parameters first, but other parts of the code relied on having a valid backend so we abort parsing the core for now and write a better warning when we encounter this situation. Good enough?

imphil, 2017-10-10:
Yeah, ignoring the whole .core file is fine and actually expected in case of a non-recoverable parse error like this one. (It's the same behavior fusesoc should have with files which happen to end in .core but are not valid core files.)

With current master it works for me, I now get:
~~~
WARNING: Unknown section 'spyglass' in '/home/no56hud/src/optimsoc/examples/fpga/vcu108/compute_tile/compute_tile_vcu108-spyglass.core'
WARNING: Parse error. Ignoring file /home/no56hud/src/optimsoc/examples/fpga/vcu108/compute_tile/compute_tile_vcu108-spyglass.core: Invalid backend "spyglass"
~~~

Thanks for the quick fix!

closed 2017-10-10 by imphil