olofk/fusesoc #179: vlogdefine not passed to verilator

lucasax, 2017-10-15:
I have noticed that a parameter passed with vlogdefine is not passed to verilator. It works fine in modelsim and icarus.

olofk, 2017-10-15:
Interesting. That is supposed to work. I'll take a look. Are you using the latest version? Do you have any examples?

olofk, 2017-10-15:
Aha! I found out why. For historical reasons, when using verilator, all parameters are sent the verilated model as command-line parameters. To get the same behaviour as for the other simulator backends, you need to set `cli_parser = fusesoc` in the verilator section.

This is really badly documented. So bad that I had forgot about it myself :/

lucasax, 2017-10-15:
It seems to work, but using or1200-generic the elf file is no longer loaded in the simulation.

olofk, 2017-10-16:
Hmm... you're right. The problem is that the file used by `--elf-load` is sent to the verilated model as a plusarg, but it expects it to be a command-line parameter. It's not trivial to fix this properly, so I recommend doing a workaround for now. The easist work-around would be to run `fusesoc sim --build-only ...` to just build the model without running it, and then launch it manually from `build/or1200_generic_0/sim-verilator/Vorpsoc_top --elf-load=/path/to/elf-file`

Another way would be to change the testbench to get the elf-file from a plusarg instead. Not sure if that also requires hacking verilator_tb_utils (this part is a bit messy unfortunately).

I need to think a bit how to improve this in a good way

lucasax, 2017-10-22:
Nice workaround, it worked perfectly!

olofk, 2017-12-05:
Can we close this?

lucasax, 2017-12-12:
Of course, sorry!

closed 2017-12-12 by lucasax