olofk/fusesoc #187: RFC: Templating Engines

wallento, 2018-01-13:
Hi,

this is a request for comments for a feature I need and want to implement. The goal is to being able to generate files during the FuseSoC run. I describe the proposed extensions below:

## Fileset

I think it makes most sense to just add file types that each correspond to an engine

    [fileset template]
    files = systemgen.sv
    file_type = template_jinja2

It may make sense to have an extension for templates, but that would add more problems in my opinions.

## Engines

I propose to support a set of python template engines and add them on-demand. [Jinja2](http://jinja.pocoo.org) seems very popular and I will build my stuff with this one.

## Flow

The normal FuseSoC flow, but instead of copying files, we feed the input + variables into the template engine and copy the output.

## Template Variables: Input

The user can supply template variables as json (or whatever you guys fancy more). The format is roughly like:

    [
      "vendor0:library0:module0": {
         "var": "value"
      }
      "vendor1:library1:module1": {
         "var": "value"
      }
    ]

The user can supply one or multiple files. All are parsed into a common variable container.

## Template Variables: Output

The template engine is called with the variables applicable to the module. It may make sense to map the other variables and parameters too, but I think we should start minimal.

## Other fancy stuff

I can think about a lot of stuff, but lets keep it simple for now.

What do you think? Would that be a good starting point? It would save me from adding an extra tool to run before fusesoc and I think it covers many use cases. I can go on and implement a prototype of that for further testing and commenting.

wallento, 2018-01-13:
This also relates to the generators by @benreynwar: https://github.com/benreynwar/fusesoc_generators

Other than those, this template engines in this RFC are deeper integrated into the flow, but obviously restrict the degrees of freedom a bit more. In my opinion they are complimentary.

imphil, 2018-01-15:
I like this proposal. Minor comments:
- Jinja2 is great. As a starting point, I've been using it here: https://github.com/imphil/fusesoc/tree/spyglass
- The common file extension for Jinja2 templates is ".j2", giving you editor highlighting in many editors.
- You need a way to specify the file type of the resulting HDL file. We have now some tools which need different code to add a SystemVerilog file vs. a Verilog file, or even special commands to add a SV-2009 vs. a SV-2001 file. I see two options:
  - Use the same file_type as today, "systemVerilog-2009" and invoke the templating logic whenever you detect a ".j2" file extension.
  - Use a combined file_type, e.g. "jinja2+systemVerilog-2009"
- I'd prefer YAML over JSON.
  - It allows adding comments to files.
  - YAML is planned to be used for CAPI2 description files, keeping the style across fusesoc the same.
- It'd be nice to be able to specify templating variables inside the .core file, as well as in a separate file. Or even merge the functionality with the params functionality that's currently present? In the end, it serves a very similar purpose.
- It also would be nice to introduce a bit of name-spacing from the start for the variables. This allows us to make pre-defined template variables available (e.g. build tool) as well as passing user-defined template variables on to the template.

benreynwar, 2018-01-15:
This sounds great!  For most of the templating I do I'd be able to use this method, and just use fusesoc_generators when the templates need access to the generic parameters of the module.

Another thing that might be nice to add is macro elaboration.

As an example, I often have a pipeline register that is actived using a generic parameter.

```
yes_use_register: if USE_THIS_REGISTER = '1' generate
    process(clk)
    begin
        if rising_edge(clk) then
            if reset = '1' then
                b_valid <= '0';
            else
                b_valid <= a_valid;
            end if;
            b_data <= a_data;
            b_blah <= a_blah;
       end if;
     end process;
end generate;

no_use_register: if USE_THIS_REGISTER = '0' generate
    a_valid <= b_valid;
    a_data <= b_data;
    a_blah <= b_blah;
end generate;
```

It's obviously tedious and bug prone to write this out all the time, and instantiating an entity doesn't work since the elements in the data flow ('data' and 'blah' in this example) will vary.  Instead I just use a macro.

```
@@ DATA_BUFFER USE_THIS_REGISTER reset a_ b_ data blah @@
```

which can be elaborated to give the above code.  Because VHDL does not support macros like this natively I expand the macro using a python function with fusesoc_generators.  It would be cool if I could define macros like this using fusesoc directly.

wallento, 2018-01-15:
Hi,

thanks for your feedback @imphil and @benreynwar. Many good points, that we should try to cover.

I started with some first drafting in the template-engines branch and a small demo of what is supported now here: https://github.com/wallento/fusesoc-templates-demo

There are a few things that need to be added, like default variable values. Other things like macro elaboration and interaction with parameters need more discussion in the following then, I think. Also about the file type I am not entirely sure right now.

Anyhow, I hope this is a good starting point for a valuable feature.

Cheers,
Stefan

benreynwar, 2018-01-16:
Thanks a bunch for working on this!  The main issue I would have using the current branch is that the template parameters must be stored in a file, and I need to create a new CoreManager every time I want to use some template parameters.

When I run my units tests on a design, I generate a whole bunch of random template parameter sets to test.  With the current implementation for every set of parameters, I'd need to write a template parameter file, instantiate a new CoreManager and then generate the files for the core.  Which would work fine, but would be a bit ugly.

It would be really cool to have functions like core.generate_files(template_parameters) to make it more pleasant to generate the files using python as opposed to through the command-line.

benreynwar, 2018-01-16:
For the macro elaboration it makes sense to wait until the templating is sorted out, since it's fairly independent.

olofk, 2018-01-16:
Thanks for starting this @wallento and great to see a constructive discussion. I would like to see a slightly different direction of this however. One issue is that I don't want any python-specific features (such as jinja2) directly in the FuseSoC codebase.

So, my proposed solution would be to have the jinja templating thing as a stand-alone generator. To elaborate a bit on the generator integration, I think I have found a good way to do this now. When a core requests a generator to be executed, the generator is responsible for creating an EDA API snippet that will be included in the main EDA API file. I already have a PoC for including EDA API yaml files in other files. This still needs the generator framework to be in place, but I think it's really time to start working on this now. The basic spec from many years ago should probably still be a good starting point.

olofk, 2018-01-16:
I made a picture that hopefully makes this a bit clearer https://www.draw.io/?lightbox=1&highlight=0000ff&edit=_blank&layers=1&nav=1&title=Untitled%20Diagram.xml#R7Vttc%2BI2EP41zNx9yfgFA%2FmYJrm2M3edzKTTXj91hC1AjSy5sgjQX9%2BVLL9JJjHEcJdL8gGstSzJu%2Fs8uyuRUXidbn8WKFt94Qmmo8BLtqPwZhQEvh9G8KUku0IynXqFYClIYjrVgnvyHzbCstuaJDhvdZScU0mytjDmjOFYtmRICL5pd1tw2p41Q0vsCO5jRF3pnySRq0I6i7xa%2Fgsmy1U5s%2B%2BZO3MUPywFXzMz3ygIF%2FqvuJ2icizTP1%2BhhG8aovB2FF4LzmVxlW6vMVW6LdVWPPdpz91q3QIz2eeBiVmG3JWvjhPQhGlyIVd8yRmit7X0J%2F16WA3gQWslUwqXPlzCnGL31ch14y%2FVuIig%2BQ%2BWcmcMjdaSg6ge%2FTPnmRkjl4I%2F4GtOudDrCT39B3cWnEkzgB%2BoCbZEfi1nhut6ruKd1Ivs1YoR5XwtYtMrMH6GxBKbXrPKIuDpmKcYXgm6CEyRJI%2Ft0ZFxuWXVzzx6JQTaNTpknDCZN0a%2BUwLoYNATzoxvGOyEvmXCw%2FrDRbGCstV4lVqk3aLbRYxeHhFdmzeOucAg8R3faXvGZkUkvs%2BQVvAGiKLtLbY9ESVLBg2KF7Iy4iMWEm%2BfNqNrIPNARSdGM0GpqU2Nab9E4qqB54m336YNvT6ptuk5kBW8DmiFLrSCbwEt34JKMH0aWk7%2Fy2hQaPkuhN4s%2FfodThJ9E%2F6dWEYfmE%2FDfXwafO98OrYUMz4jnY47teZf4ARd7EAHtuogscrUZbyjBHQontffvFD25%2Fn5FBpNukN3S6Neh0ZnA2jUd%2F3tBOQTnoR8jqOYyGWY8QsZpq%2Byo073DS6WmGGBJBev3ZO9tif7U9eTJydy5NBzdHYCR55%2BR448G96RjwqVUXBYqeL0n0wGDa0zB2WjYIJSBQ82zzOt71O3vWsA9hfE0BKgug%2FLucSqd4YFgddTmNaiu7r9bLQnW1xunJwH4nadGXRAvDP8D4Hxy31JU%2FjKkqYwOmPS5J%2BHGV9HgXHpkuYgRahbNU4toESWKYtVmaeeoMvApksvag9UvIsz0MFl8aU1z%2ByZstjuHwxcFrsl0vl53OH1OlX7oWjdJqiuzO1ktO6Pz0FQp%2BWnYvyKoYbcJCuAMDxBWcl65QKHEpSd9VcBZ2iC2jPPYITjphd7Mb6geHuljpi0jyXm8iamKM9JbKOZ0oYrmZOgU9QHro804Bt1oLeUvdCVquLLWGZss0LvWGft1Y97utIxBxqeY229lwSiBGdgU%2FBEj7OROl6EGl2tgSWm5aabwLyybfW2dRln2HIFIyq5PAZT6tigeJzEiF6ZGylJEs14XUGjzYLdfvOiwGBH%2BXHX6Ynf4Vo2mxwTGEoAviPyYHK3gOTbQOqLSN%2Fe7vVOh8juPfFCG%2F%2Buca6Cg4eaSZiKlBw%2B5qoIBMxQnJSiWGAksY6uqfpieAOfYHmcvx30RhZ4e2J3iKqzY6%2F%2BHbu9sDu2Cj7fhlxf7I6tsOzbYXlA7Lpb281KyaBR4VeuHDSayIpUsL29uYLPq7tflacwkmVY6oeQ1IE5jwWZl%2BOk8PVBjaGaO%2FCkANbpUR6D9lXkVi0s448X8P0b12RgBqrWlmveUBMjmrdo4w%2B1hE83eiGUzPNiOHASpIu4vBhbD0lUY4N2b4dXnF1Al1iCrmpxOgSxTN6J5ThimQy1JWWfWDq7AAMSy7QzKWgclHmExXSdNMiFw5dKDYrbiiAUA9wSI54rjeeAekbYsnpow8XD3%2FondxrovOqpEo4UxyvESJ7WFELYgou0nrS14e8BKuNVg%2BF0SmIW%2BnZoYmLThB%2B5CUjXgeAgPNF1DvTOEz14IrLLeTsf7M0T1kChbdcBeaL7bCitEGlnIM1Ew%2BQQSN1d52vkHr%2F%2FsBAdT6O2rbs2fk9V34f%2BO0SPg2hob8sE1qFQX4iGVk4QRv1Ol475yZu7m%2FO7hmKx34ZZrPQnBVYynVMvdDkPM%2FMKtm1QF4WDwDmnj3Xpj3S%2FdE6YloHV17Fc64PiDxXsP74djIfWHk5VFz6H8cvDMQ7N%2Bv8WCiep%2FzkkvP0f

olofk, 2018-10-08:
With the added support for FuseSoC generators, I recommend implementing this functionality in a generator instead

closed 2018-10-08 by olofk