olofk/fusesoc #189: NameError: name 'parser' is not defined

imphil, 2018-01-18:
A up-to-date fusesoc checkout (master) gives the following error when called without any command line arguments:

~~~
philipp@powerashley:~/src/fusesoc> ~/.local/bin/fusesoc
Traceback (most recent call last):
  File "/home/philipp/.local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/home/philipp/src/fusesoc/fusesoc/main.py", line 488, in main
    parser.print_help()
NameError: name 'parser' is not defined
~~~

olofk, 2018-01-18:
Thanks! Fixed (reopeon if I'm wrong)

closed 2018-01-18 by olofk

imphil, 2018-01-18:
Fixed, thanks.

tianbartek, 2018-12-07:
I have the same problem. How to fix? Can you be more specific

imphil, 2018-12-07:
@olofk fixed this in https://github.com/olofk/fusesoc/commit/8c06ad7d369672ca3d9fc68c09e417faf97d44b2. If you experience the same or a similar problem, please make sure that you're running the latest fusesoc version. If that's the case, please open a new issue, since yours is likely different from mine (which was fixed).

MPrakash007, 2020-02-01:
data = Parser.parse_args()
NameError: name 'Parser' is not defined

carminestore, 2021-07-13:
Can someone please help me..i Have the same error

imphil, 2021-07-13:
@carminestore can you please open a new issue and include the version of fusesoc you're using, more information about your system (Python version, etc.), and the command you're trying to run?

alessiobartels98, 2021-08-05:
parsed = parser.from_file('analisi.pdf')
#print(parsed['content'])
pdf = parsed["content"]
pdf = pdf.split("\n")
#print(len(pdf))
print(pdf)

name Error: name 'Parser' is not defined

alessiobartels98, 2021-08-05:
can someone help me please?

imphil, 2021-08-05:
This does not look like fusesoc code?