olofk/fusesoc #195: Do not write fusesoc.log by default

imphil, 2018-02-28:
Fusesoc by default writes a file 'fusesoc.log' into the working directory. Even though this is common for many EDA tools, I suggest removing this functionality by default.

Why?
- Having two fusesoc runs in the same directory results in an "interleaved" fusesoc.log file, essentially a useless one.
- It's very uncommon behavior for any "non-EDA" tool.

Two ideas how to resolve this:
- Write the fusesoc.log file to the build directory (`build/DESIGNNAME/...`)
- Don't write a fusesoc.log file unless instructed so by e.g. a command line option (e.g. `--logfile=fusesoc.log`)

olofk, 2018-02-28:
A thousand times yes!! It has annoyed me so many times. Especially since I hate when other EDA tools leave unnecessary files around. I just never got down to fixing it. Just adding an optional command-line option is enough I would say

wallento, 2018-02-28:
A first mitigation is to add the pid to the filename or so, but overall I am the third in the gang of haters of tools writing files without me asking for it..

olofk, 2018-02-28:
Yeah, just leave it to the user to specify a unique filename if they need that

commit 89e74b9e (2018-03-03): Do not write fusesoc.log file by default

With a new option --log-file users can specify a log file if they want
to. Only in this case a log is written, otherwise all output is only on
STDOUT.

Fixes #195
closed 2018-03-03 by olofk