olofk/fusesoc #196: Build sockit fails

wtfuzz, 2018-03-01:
```
Info (12128): Elaborating entity "mor1kx_wb_mux_cappuccino" for hierarchy "mor1kx:mor1kx0|mor1kx_cpu:mor1kx_cpu|mor1kx_cpu_cappuccino:cappuccino.mor1kx_cpu|mor1kx_wb_mux_cappuccino:mor1kx_wb_mux_cappuccino" File: /home/fuzz/build/sockit_0/src/mor1kx_5.0-r1/rtl/verilog/mor1kx_cpu_cappuccino.v Line: 1055
Info (12128): Elaborating entity "mor1kx_rf_cappuccino" for hierarchy "mor1kx:mor1kx0|mor1kx_cpu:mor1kx_cpu|mor1kx_cpu_cappuccino:cappuccino.mor1kx_cpu|mor1kx_rf_cappuccino:mor1kx_rf_cappuccino" File: /home/fuzz/build/sockit_0/src/mor1kx_5.0-r1/rtl/verilog/mor1kx_cpu_cappuccino.v Line: 1125
Error (10198): Verilog HDL error at mor1kx_rf_cappuccino.v(283): part-select direction is opposite from prefix index direction File: /home/fuzz/build/sockit_0/src/mor1kx_5.0-r1/rtl/verilog/mor1kx_rf_cappuccino.v Line: 283
Info (10499): HDL info at mor1kx_rf_cappuccino.v(283): see declaration for object "wb_rfd_adr_expand" File: /home/fuzz/build/sockit_0/src/mor1kx_5.0-r1/rtl/verilog/mor1kx_rf_cappuccino.v Line: 283
Error (12152): Can't elaborate user hierarchy "mor1kx:mor1kx0|mor1kx_cpu:mor1kx_cpu|mor1kx_cpu_cappuccino:cappuccino.mor1kx_cpu|mor1kx_rf_cappuccino:mor1kx_rf_cappuccino" File: /home/fuzz/build/sockit_0/src/mor1kx_5.0-r1/rtl/verilog/mor1kx_cpu_cappuccino.v Line: 1125
```

wtfuzz, 2018-03-01:
Using Quartus 17.1

olofk, 2018-03-02:
Technically, I would like to have issues with the different cores filed against the core library (i.e. orpsoc-cores). No big deal. Just mentioning for the future.

I think the problem is a bad commit in mor1kx that causes parse errors for some tools. As there is no explicit version set in the dependency, it will try to use the latest mor1kx. Could you please try to change this line https://github.com/openrisc/orpsoc-cores/blob/master/systems/sockit/sockit.core#L10 to `mor1kx-4.1` instead and see if that works. That would force it to use a known good version of mor1kx.

You will find the file in ~/.local/share/fusesoc/orpsoc-cores/systems/sockit/sockit.core

wtfuzz, 2018-03-02:
I tried with mor1kx-3.1 yesterday and it worked.

I think the problem is in mor1kx. I've been working on building a fusesoc core for a Terasic C5G, and discovered that mor1kx-5.0-r1 will only compile if I enable shadow registers.

I haven't had too much time to dig into this or try other versions, as I've been busy getting an LPDDR2 controller wired up.

olofk, 2018-03-06:
Yes, you are right. The problem is a bad commit for mor1kx that made some tools unable to parse the code.  There is a proposed fix, but no one has had time to look at that yet unfortunately. As a workaround, I suggest using a slightly older mor1kx in the meantime.

Also good to know about the problem with shadow registers. I will make sure that we test both with and without next time we update the repo.

olofk, 2020-02-29:
Closing due to inactivity

closed 2020-02-29 by olofk