olofk/fusesoc #204: --config option does not resolve "~" in path names

ThomasHornschuh, 2018-04-01:
I added a local fusesoc.conf file with the --config option to my fusesoc command line. This is the content of the file:
````
[main]
cores_root =~/development/bonfire
build_root = ~/development/Vivado/build/2017.4
cache_root = ~/.cache/fusesoc

````
fusesoc build did not resolve the  ` ~`   to my home directory, instead it created a directory under the cwd with the name "~" to add the build and .cache directories.

Maybe this is not supported, but I think I remember that the default config created upon installation contains at least for cache_root  exactly the path above (I cannot check, I have no clean install anymore :-)

olofk, 2018-04-29:
This should be fixed now with 4d12e82 . Please close if this fixes the issue

ThomasHornschuh, 2018-06-01:
Works...

closed 2018-06-01 by ThomasHornschuh