olofk/fusesoc #206: fusesoc fetch is not fetching

bandvig, 2018-04-22:
I have installed FuseSoC and done "fusesoc init" with default paths for fusesoc.conf and core libraries. Than I tried executed "fusesoc --verbose fetch de0_nano", it looks like it did nothing. I mean it did not fetch. Here is the log:
$ cd ~/ORPSoC/de0_nano_build
$ fusesoc --verbose fetch de0_nano
DEBUG: Setup logging at level 10.
DEBUG: Verbose output
DEBUG: Colorful output
DEBUG: Looking for config files from /etc/fusesoc/fusesoc.conf:/home/BAndViG/.config/fusesoc/fusesoc.conf:fusesoc.conf
DEBUG: Found config files in /home/BAndViG/.config/fusesoc/fusesoc.conf
DEBUG: build_root=/home/BAndViG/ORPSoC/de0_nano_build/build
DEBUG: cache_root=/home/BAndViG/.cache/fusesoc
DEBUG: cores_root=
DEBUG: Not defined
DEBUG: library_root=/home/BAndViG/.local/share/fusesoc
DEBUG: Command line arguments: ['/bin/fusesoc', '--verbose', 'fetch', 'de0_nano']
DEBUG: Checking for cores in /home/BAndViG/.local/share/fusesoc/orpsoc-cores
DEBUG: Adding core ::ac97:0
DEBUG: Adding core ::adv_debug_sys:0
DEBUG: Adding core ::altera_virtual_jtag:0
DEBUG: Adding core ::cfi_ctrl:0
DEBUG: Adding core ::csrbrg:0
DEBUG: Adding core ::diila:0
DEBUG: Adding core ::ethmac:0
DEBUG: Adding core ::s25fl064p:1.7
DEBUG: Adding core ::gpio:0
DEBUG: Adding core ::i2c:1.13
DEBUG: Adding core ::i2c:1.14
DEBUG: Adding core ::i2c:0
DEBUG: Adding core ::jtag_tap:1.13
DEBUG: Adding core ::jtag_tap:0
DEBUG: Adding core ::jtag_vpi:0-r2
DEBUG: Adding core ::libaxis:1.0
DEBUG: Adding core ::libstorage:1.0
DEBUG: Adding core ::minimac2:0
DEBUG: Adding core ::mmuart:0
DEBUG: Adding core ::mor1kx:3.1
DEBUG: Adding core ::mor1kx:4.1
DEBUG: Adding core ::mor1kx:0
DEBUG: Adding core ::mt46v16m16p:0
DEBUG: Adding core ::mt48lc16m16a2:0
DEBUG: Adding core ::nyuzi:0
DEBUG: Adding core ::opa:0
DEBUG: Adding core ::or1200:0-r2
DEBUG: Adding core ::or1200:0
DEBUG: Adding core ::or1k_bootloaders:0.9
DEBUG: Adding core ::osvvm:2015.06
DEBUG: Adding core ::picorv32:0
DEBUG: Adding core ::ps2:0
DEBUG: Adding core ::apb_event_unit:0
DEBUG: Adding core ::apb_ffl_if:0
DEBUG: Adding core ::apb_gpio:0
DEBUG: Adding core ::apb_i2c:0
DEBUG: Adding core ::apb_pulpino:0
DEBUG: Adding core ::apb_spi_master:0
DEBUG: Adding core ::apb_timer:0
DEBUG: Adding core ::apb_uart:0
DEBUG: Adding core ::axi2apb:0
DEBUG: Adding core ::axi_mem_if_DP:0
DEBUG: Adding core ::axi_node:0
DEBUG: Adding core ::axi_slice:0
DEBUG: Adding core ::axi_slice_dc:0
DEBUG: Adding core ::axi_spi_master:0
DEBUG: Adding core ::axi_spi_slave:0
DEBUG: Adding core ::pulpino:0
DEBUG: Adding core ::pulpino_adv_debug_sys:0
DEBUG: Adding core ::ri5cy:0-r1
DEBUG: Adding core ::ri5cy:0
DEBUG: Adding core ::ridecore:0
DEBUG: Adding core ::scalable_arbiter:0
DEBUG: Adding core ::SD-card-controller:0-r1
DEBUG: Adding core ::SD-card-controller:0
DEBUG: Adding core ::simple_spi:1.6
DEBUG: Adding core ::simple_spi:0
DEBUG: Adding core ::uart16550:1.5.2
DEBUG: Adding core ::uart16550:1.5.3
DEBUG: Adding core ::uart16550:1.5.4
DEBUG: Adding core ::uart16550:1.5
DEBUG: Adding core ::uart16550_model:0
DEBUG: Adding core ::uvvm_util:1.2.0
DEBUG: Adding core ::verilator_tb_utils:0
DEBUG: Adding core ::verilog-arbiter:0-r1
DEBUG: Adding core ::verilog-arbiter:0
DEBUG: Adding core ::verilog_utils:0
DEBUG: Adding core ::vga_lcd:0
DEBUG: Adding core ::vscale:0
DEBUG: Adding core ::wb_avalon_bridge:0
DEBUG: Adding core ::wb_bfm:1.0
DEBUG: Adding core ::wb_bfm:0
DEBUG: Adding core ::wb_common:0
DEBUG: Adding core ::wb_ram:1.0
DEBUG: Adding core ::wb_riscvscale:0
DEBUG: Adding core ::wiredelay:0
DEBUG: Adding core ::xilibs:0
DEBUG: Adding core ::atlys:0
DEBUG: Adding core ::de0_nano:0
DEBUG: Adding core ::de1:0
DEBUG: Adding core ::de2:0
DEBUG: Adding core ::mor1kx-generic:0
DEBUG: Adding core ::neek:0
DEBUG: Adding core ::nyuzi_de2_115:0
DEBUG: Adding core ::or1200-generic:0
DEBUG: Adding core ::sockit:0
DEBUG: Adding core ::vscale-generic:0
DEBUG: Checking for cores in /home/BAndViG/.local/share/fusesoc/fusesoc-cores
DEBUG: Adding core ::ac97:1.2
DEBUG: Adding core ::adv_debug_sys:3.1.0-r1
DEBUG: Adding core ::adv_debug_sys:3.1.0
DEBUG: Adding core ::altera_virtual_jtag:1.0
DEBUG: Adding core ::cdc_utils:0.1
DEBUG: Adding core ::ddr2_1024:5.83
DEBUG: Adding core ::de0_nano-multicore:1.0
DEBUG: Adding core ::elf-loader:1.0.1
DEBUG: Adding core ::elf-loader:1.0.2
DEBUG: Adding core ::elf-loader:1.0
DEBUG: Adding core ::fifo:1.0
DEBUG: Adding core ::fifo:1.1
DEBUG: Adding core ::fifo:1.2.1
DEBUG: Adding core ::fifo:1.2
DEBUG: Adding core ::gpio:1.0
DEBUG: Adding core ::i2c:1.14-r1
DEBUG: Adding core ::i2c:1.14
DEBUG: Replacing ::i2c:1.14 in /home/BAndViG/.local/share/fusesoc/orpsoc-cores/cores/i2c with the version found in /home/BAndViG/.local/share/fusesoc/fusesoc-cores/i2c
DEBUG: Adding core ::jtag_tap:1.13-r1
DEBUG: Adding core ::jtag_tap:1.13
DEBUG: Replacing ::jtag_tap:1.13 in /home/BAndViG/.local/share/fusesoc/orpsoc-cores/cores/jtag_tap with the version found in /home/BAndViG/.local/share/fusesoc/fusesoc-cores/jtag_tap
DEBUG: Adding core ::jtag_vpi:0-r2
DEBUG: Replacing ::jtag_vpi:0-r2 in /home/BAndViG/.local/share/fusesoc/orpsoc-cores/cores/jtag_vpi with the version found in /home/BAndViG/.local/share/fusesoc/fusesoc-cores/jtag_vpi
DEBUG: Adding core ::mor1kx:5.0-r1
DEBUG: Adding core ::mor1kx:5.0-r2
DEBUG: Adding core ::mor1kx:5.0
DEBUG: Adding core ::n25q128a11e:1.2
DEBUG: Adding core ::ompic:1.0
DEBUG: Adding core ::or1k_bootloaders:0.9.1
DEBUG: Adding core ::simple_spi:1.6
DEBUG: Replacing ::simple_spi:1.6 in /home/BAndViG/.local/share/fusesoc/orpsoc-cores/cores/simple_spi with the version found in /home/BAndViG/.local/share/fusesoc/fusesoc-cores/simple_spi
DEBUG: Adding core ::stream_utils:1.1
DEBUG: Adding core ::stream_utils:1.2
DEBUG: Adding core ::stream_utils:1.3
DEBUG: Adding core ::timer:1.0
DEBUG: Adding core ::uart16550:1.5.5-r1
DEBUG: Adding core ::uart16550:1.5.5
DEBUG: Adding core ::verilog-arbiter:0-r1
DEBUG: Replacing ::verilog-arbiter:0-r1 in /home/BAndViG/.local/share/fusesoc/orpsoc-cores/cores/verilog-arbiter with the version found in /home/BAndViG/.local/share/fusesoc/fusesoc-cores/verilog-arbiter
DEBUG: Adding core ::verilog-arbiter:0-r2
DEBUG: Adding core ::vlog_tb_utils:1.1
DEBUG: Adding core ::wb_altera_ddr_wrapper:0-r1
DEBUG: Adding core ::wb_bfm:1.2.1
DEBUG: Adding core ::wb_bfm:1.2
DEBUG: Adding core ::wb_common:1.0.1
DEBUG: Adding core ::wb_common:1.0.2
DEBUG: Adding core ::wb_common:1.0
DEBUG: Adding core ::wb_intercon:1.1
DEBUG: Adding core ::wb_intercon:1.2.1
DEBUG: Adding core ::wb_intercon:1.2
DEBUG: Adding core ::wb_ram:1.1
DEBUG: Adding core ::wb_sdram_ctrl:0-r2
DEBUG: Adding core ::wb_streamer:1.1

olofk, 2018-04-22:
Hi Andrey,

The reason why nothing happens is that the de0_nano core is already on the disk. It's likely in `~/.local/share/fusesoc/orpsoc-cores/systems/de0_nano`. If it's not there, you can run `fusesoc core-info de0_nano` and look for `Core root` to see where to find the code. Hope this helps

bandvig, 2018-04-24:
I tried `fusecoc build de0_nano-multicore`. It cached all dependancies normally (into `~/.cache/fusesoc`), copied them into `./build/de0_nano-multicore_1.0/src` and tried to build. As I haven't got Quartus FuseSoC stopped with error of course. However all pre-building steps were successfull. 
I didn't try `fetch` command before `build` at that time. Could it be reason for difference `de0_nano` from `de0_nano-multicore`?

**$ fusesoc build de0_nano** command still reports conflicting requirements.

bandvig, 2018-04-27:
Thanks, Olof. Removing `FUSESOC_IGNORE` and using `fusesoc build --setup` works.

baloncu, 2018-04-30:
Hi, 
I am directed to this thread from a recent one I started in opencores.org. 
I went ahead and removed the FUSESOC_IGNORE files, it seems to download everything and setup for de0_nano completes with the following warnings. I plan to ignore them until there is another problem.  thank you :)

WARNING: ::wb_intercon:1.0 : plusargs section is deprecated and will not be pars                                     ed by FuseSoC. Please migrate to parameters
WARNING: ::wb_intercon:0 : plusargs section is deprecated and will not be parsed                                      by FuseSoC. Please migrate to parameters

olofk, 2020-02-29:
Seems like it all worked out eventually. Closing this

closed 2020-02-29 by olofk