olofk/fusesoc #207: Enable modelsim GUI option

lvoudour, 2018-04-29:
Hi,
At the moment modelsim simulations are run by default in command line mode using the `-c` switch which is hard-coded in the run target of modelsim.py. Specifying the `-gui` or `-i` switches in` vsim_options` of the .core file doesn't work because vsim complains that the two switches are mutually exclusive.

Removing the `-c` switch from the run target of modelsim.py would fix the problem (it will still run in command line by default) allowing users to start a gui simulation using `vsim_options`

The only problem is I don't know if this works in older versions of modelsim. I'm using the 10.5b Intel edition but I remember some older versions had different behavior.

I could make a pull request if you are fine with it

olofk, 2018-04-29:
Hi,

Starting modelsim in GUI mode is something I want to support, but I see two problems with dropping `-c` and setting `-gui` in vsim_options. First, modelsim starts up in GUI mode when I drop the `-c` argument, which is strange since I'm also using 10.5b intel edition. Not sure what's different between our setups, but this indicates that CLI mode might break for other cores by dropping `-c`. The other thing is that you might want to run regression tests in CLI mode and do some debugging in GUI mode, and this will require to edit `vsim_options` every time you want to switch. Also in addition to that, I want to make it possible to launch all tools in GUI mode.

I have an alternative solution instead, that is done in two steps. First of all, the ModelSim Makefile should be updated to add a new `run-gui` target. I have already done that for the `isim` and `xsim` backends. When that is completed, it's possible to run `fusesoc run --setup --target=<target> <core>` to just create the project files, and then run `make run-gui` from the work root (e.g. `build/your_core/<target>-modelsim`). Then I want to add a `--gui` flag to FuseSoC that will call the `run-gui` target in the run stage of FuseSoC (or in the build stage for synthesis tools).

So I would be happy to have a PR that adds a `run-gui` target to the modelsim makefile (as done in isim and xsim), and I would also be happy for a PR that adds a --gui flag and the necessary logic to run supported tools in GUI mode.

lvoudour, 2018-04-29:
> I have an alternative solution instead, that is done in two steps. First of all, the ModelSim Makefile should be updated to add a new run-gui target. I have already done that for the isim and xsim [...]

Yes that's much cleaner, just tested works like a charm

> First, modelsim starts up in GUI mode when I drop the -c argument, which is strange since I'm also using 10.5b intel edition. Not sure what's different between our setups, but this indicates that CLI mode might break for other cores by dropping -c

Strange, I tested again without the `-c` before pulling and it doesn't invoke the gui. I think it doesn't run in gui when there's a `-do 'tcl stuff'` option but maybe I'm wrong and there's some option set in my modelsim.ini. Anyway no worries it's better the way you've changed it. Cheers!

closed 2018-04-29 by lvoudour