olofk/fusesoc #212: CAPI2: Rename "toplevel" to plural form

imphil, 2018-05-19:
CAPI2 files expect in the "toplevel" key an array of strings representing toplevel modules. An example can be found here: https://github.com/optimsoc/optimsoc/blob/spyglass/examples/fpga/nexys4ddr/compute_tile/compute_tile_nexys4ddr_capi2.core#L52

That's counterintuitive in a couple of ways:
- Passing a string makes the parser create a list of individual letters, certainly not what people expect.
- The key name "toplevel" indicates a single value, while a list is expected.

I'm therefore proposing to rename the key to "toplevels". This makes it clear that a list of elements is expected. 

Another option would be to make toplevel expect really a string (actually the most common case), and have a separate option "toplevels" with the ability to specify multiple toplevel modules.

olofk, 2018-05-20:
I have thought about this too. It's a bit awkward since, as you say, the most common case is a single toplevel. The idea I have been planning to implement is to simply allow toplevel to be either a string or a list. That will make it backwards compatible, and I don't think it will really cause any problems.

olofk, 2018-05-20:
Could you try [this](https://github.com/olofk/fusesoc/tree/toplevel-string) and see if that works for you?

imphil, 2018-05-21:
The code in toplevel-string works.

olofk, 2018-05-21:
Fixed with  7901f79

closed 2018-05-21 by olofk