olofk/fusesoc #217: Quartus backend support for compiling into multiple libraries

GCHQDeveloper992, 2018-07-06:
Hello! I'm currently experimenting with FuseSoC to see if I can make use of it to build some of our projects. I've noticed that the Quartus backend does not presently look for a logical_name associated with a file/fileset, meaning that projects which place files into separate libraries and then explicitly reference them fail to compile.

I have a local branch that I'm using to experiment with this functionality, which I could put up as a Pull Request if that would be helpful?

olofk, 2018-07-08:
Oh, you're right. That's definitely a bug and I would happily accept patches to have that fixed. Thanks for reporting and just throw me a PR when you're ready

GCHQDeveloper992, 2018-07-09:
Just a thought after I pushed this - would it be nicer to dynamically create the TCL string based on a dictionary? For example, whilst parsing each file in the for loop from Line 95 we could add entries into a dictionary (like {"-library" : "libx", "-name": "vhdl_lfile"}) and then after populating this structure, loop over the key/value pairs to construct the actual TCL string. That way, we could avoid the if condition I've added in this PR and (if another option needed adding in the future) avoid deeply nested if statements. 

Is this something you'd like, or should we leave it for now?

olofk, 2018-07-09:
Thanks for the PR. It's pushed now. I think what you propose is a nicer solution, but we can skip that for now as I want to make bigger changes for most backends and use more templates, as it is done in the Vivado backend.

closed 2018-07-09 by olofk